gyroflow/gyroflow #162: Insta360 Go (v1) - Profiles for lens and stab

nightflasher, 2022-02-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0-rc2

### What operating system are you using?

ubuntu 20.04.3 LTS

### What GPU are you using?

Radeon

### What happened?

I have seen there are two or three lens profiles, but they really  don't fit the Go (v1). If i try it anyways, there is a bit of lens correction (seems to be fine for the moment), but i can't get any stabilized footage.
The footage is from my 3' Donut Cinewhoop and it's quite stable ootb - so...something very strange happening or nothing good at all.

kwkwkwkw, 2022-02-14:
I also tested with the GO v1 and have the same problem for the lens correction with the 2 profiles in the list.  It seems too wide for the original footage and will makes the usable area to be too small for output.  So I tried to make my own profile with the output from the insta360 studio using their converted footage (make it 16:9 2720 x 1530 instead of square 1:1 2720 x 2720).

It's way better with this lens correction profile than the 2 listed one but still I guess RC3 is not support the GO v1's gyro data in the original footage file (extension .insv) although it can read and show the gyro graph at the bottom area.  It seems doesn't make any effect on the stabilization process.  As read from the supported gyro source in RC3 and the telemetry-parser, both are not support the Insta360 GO v1's gyro data.

I really wish in the soon future the new release will make the GO v1 being compatible as I think there are a lot of pilots are still using this action camera.

AdrianEddy, 2022-02-15:
Can you upload any test file from Go1?

nightflasher, 2022-02-15:
i've uploaded a short clip: https://www.dropbox.com/s/feyrsyuml7716km/VID_20200710_132023_00_011.insv?dl=0

nightflasher, 2022-02-15:
1. Which software to use for lens profile? (Too lazy to look for of there is documentation about it)
2. I'm not getting stabilization - is this fixed with next release?

AdrianEddy, 2022-02-15:
1. Gyroflow https://docs.gyroflow.xyz/guide/calibration/
2. It's fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-02-22 by AdrianEddy