gyroflow/gyroflow #166: Win7 support

StuweFPV, 2022-02-11:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I know the system requirements say Win10 but I'm still on Win7 and would love to use that tool. Hard to understand why this tiny little tool requires Win10 when I run Inav, Adobe Suite, Insta360 Suite, GoPro, Filmora, etc. all on Win7.
I did install the distribution pack but no luck. He's always complaining about OpenCL.dll missing. 
Could you please add legacy OS compatibility as not the whole world is always on the latest OS. 
thanks

b14ckyy, 2022-02-11:
It was answered already 
https://github.com/gyroflow/gyroflow/issues/82

AdrianEddy, 2022-02-11:
What is the reason to still use Win7 in 2022? Microsoft stopped officially supporting it 2 years ago

StuweFPV, 2022-02-12:
> It was answered already #82

Thanks. I did do a search on windows 7 and win7 but no luck

closed 2022-02-12 by StuweFPV