gyroflow/gyroflow #172: Gyroflow crashes when importing Gopro H265 Footage

gabriel3504, 2022-02-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

Gyroflow 1.0.0 rc3

### What operating system are you using?

Windows 11

### What GPU are you using?

Nvidea MX150

### What happened?

I tried multiple H265 Gopro files, when I import them, Gyroflow crashes

### Relevant log output

```shell
09:04:52 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\gabri\\Desktop\\Gyroflow-windows64\\camera_presets"
09:04:52 [INFO] Loaded 1290 lens profiles in 37.324ms
09:04:52 [INFO] Latest version: v1.0.0-rc3, current version: 1.0.0-rc3
09:04:52 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce MX150
09:04:52 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
09:04:53 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src\rendering\mod.rs:45] gpu_type = NVIDIA
09:04:57 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
09:04:57 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.video" ""
09:04:57 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
09:04:57 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
09:04:57 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
09:04:57 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
09:04:57 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x2346ac4b2a0) QSize(873, 655)
09:04:58 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.video" "MFT"
Drücken Sie eine beliebige Taste . . .
```

AdrianEddy, 2022-02-15:
Please test with latest dev build and upload gyroflow.log afterward here https://gyroflow.xyz/devbuild/?autodownload

gabriel3504, 2022-02-15:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8068607/gyroflow.log)

AdrianEddy, 2022-02-15:
@wang-bin

wang-bin, 2022-02-15:
try [the latest sdk](https://sourceforge.net/projects/mdk-sdk/files/nightly/mdk-sdk-windows-desktop-vs2022.7z/download), and upload a log if still crash

AdrianEddy, 2022-02-15:
@gabriel3504 so download the latest mdk-sdk from the link provided, and copy mdk-sdk\bin\x64\mdk.dll file to Gyroflow folder (replace mdk.dll there), try again and post log if crashed again

gabriel3504, 2022-02-15:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8073902/gyroflow.log)
Still crashing...

AdrianEddy, 2022-02-15:
Stack trace is probably this:
```
[0x0]   mdk!mdk::abi::NativeVideoBufferPool::create + 0x10931
[0x1]   mdk!mdk::abi::D3D11Video::render + 0x47
[0x2]   mdk!mdk::abi::D3D11Video::set + 0x1b75
[0x3]   mdk!mdk::abi::VideoRenderer::redraw + 0x9f6
[0x4]   mdk!mdk::abi::VideoRenderer::draw + 0x57f
[0x5]   mdk!mdk::abi::Player::renderVideo + 0x33
[0x6]   gyroflow + 0x6ae08c
....
```

wang-bin, 2022-02-16:
> Stack trace is probably this:
>
> ```
> [0x0]   mdk!mdk::abi::NativeVideoBufferPool::create + 0x10931
> [0x1]   mdk!mdk::abi::D3D11Video::render + 0x47
> [0x2]   mdk!mdk::abi::D3D11Video::set + 0x1b75 [...]

Seems pdb and dll do not match.

From the log

>  19:05:02 [DEBUG] (13) mdk: [MDK] input D3D11_TEXTURE2D_DESC.Width: 4000, Height: 3000, MipLevels: 1, ArraySize: 1, Format: 103, SampleDesc{1, 0}, Usage: 2, BindFlags: 208, CPUAccessFlags: 10000, MiscFlags: 0. desired bindFlags: 8, miscFlags: 0create a shared texture for input to copy input texture
> 19:05:02 [DEBUG] (13) mdk: [MDK] dev_->CreateTexture2D(&td, nullptr, &tex2d_)  ERROR@55D3D11::Texture::ensureOnDevice: (80070057) Falscher Parameter.

The texture from decoder can be written by cpu and read by gpu, usually CPUAccessFlags is 0, i guess the reason is hardware accelerated decoding is not supported by gpu, video is decoded by cpu and write to texture for gpu. There are other proofs show MX150 does not support video decoding and encoding, see https://forums.developer.nvidia.com/t/mx150-seems-not-to-support-nvdec-nvenc-in-windows-10/77628  and https://developer.nvidia.com/video-encode-and-decode-gpu-support-matrix-new .  You can also use DXVAChecker to find out supported decoders. A screenshot will be nice.

Please download the latest sdk again, replace the dll and pdb, test and upload the log. `CreateTexture2D()` error may disappear, if still exists, it will fallback to cpu read.

AdrianEddy, 2022-02-16:
Ok so this should be fixed in latest dev build: https://gyroflow.xyz/devbuild/?autodownload (v1.0.0-rc3 (gh289)). Please test and confirm

gabriel3504, 2022-02-16:
IT WORKS! Thank you so much for all the support and troubleshooting

AdrianEddy, 2022-02-16:
🥳

closed 2022-02-16 by AdrianEddy