gyroflow/gyroflow #177: Right click should not move timeline cursor

Seba112PL, 2022-02-13:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

When right clicking on the timeline timeline indicator moves to that spot. When doing manual syncing this is not a good option. Im using next frame, previous frame to go to spot that i think will be the best but then when I right click it changes position of cursor.

relaxgit, 2022-02-14:
Yes, moving the timeline indicator with left button is enough. If i right click to place a manual sync point, position shouldn't move at all. Now it's very easy to move accidentally.

Also it would be nice to move the indicator with dragging the whole vertical line, not just the top of it. And the cursor should change to a small left-right arrow in this situation.

AdrianEddy, 2022-02-15:
Fixed in 1a9c1a737ad480d313128a8ba1ea7aa6043ac7c7

closed 2022-02-15 by AdrianEddy

relaxgit, 2022-02-16:
timeline cursor doesn't move now, but the sync point wont be generated at that place, instead at the current mouse position

AdrianEddy, 2022-02-17:
I changed it now so that right click doesn't move the playhead, but only when clicked 100px around the playhead.
If you right click in a completely different timeline area, it will move the playhead there, I think that's more intuitive and it still allows you to have the precision when clicking around current location