gyroflow/gyroflow #178: Previous frame not working

Seba112PL, 2022-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

rc3 gh257

### What operating system are you using?

Windows 10

### What happened?

Next frame is working fine but previous not.

[image]

lukegluke, 2022-05-13:
Still the same in 1.0.1. Previous frame (left) works only couple frames, than stops, while long jump (cntrl+left) works better.

AdrianEddy, 2022-06-10:
This is essentially the same as #296

closed 2022-06-10 by AdrianEddy

AdrianEddy, 2022-06-20:
Fixed in latest dev build