gyroflow/gyroflow #179: Quality drops on export from Gyroflow vs regular or trough ofx

xtrul, 2022-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0 rc 3

### What operating system are you using?

windows 11

### What GPU are you using?

gtx 1070ti

### What happened?

The issue is, that when exported from the app, no matter what format is h264,hevc, prores, png quality drops visible!

You can check this youtube clip as rought comparation(also have a download link for better quality)

https://youtu.be/aUjpeW9zt3w

Thank you
Sebastian

AdrianEddy, 2022-02-13:
That is weird, ofx uses exactly the same code, and png is not compressed so it's neither interpolation nor compression, so what's going on?

Entropy512, 2022-02-17:
@xtrul out of curiosity, when using gyroflow as an OFX plugin:  What host software?  Resolve?

If so - are you going through Fusion?  What colorspace conversions (if any) - is there a possibility the OFX plugin is working on linear data instead of gamma-compressed?

xtrul, 2022-02-17:
@Entropy512 hi i was testing on Resolve, yes. My workflow was adding a "raw stream of dngs" exporting as a video file, making the gyro corrections, exporting from there the settings file, back into the original dng stream, Fursion new node "ofx plugin" loading the corrections , after i was doing any other node effects and exporting

Entropy512, 2022-02-17:
Hmm...  So when exporting from the app, your input file is an exported video instead of the DNG steam?  (I'm guessing your camera records CinemaDNG?)

Do you happen to know if the colorspace is linear for the input and output of the OFX plugin?

I pointed @AdrianEddy to http://www.ericbrasseur.org/gamma.html?i=1 as a possible root cause of this issue in a discussion elsewhere.  (We should probably stop polluting that discussion with stuff related to this issue.)

@AdrianEddy the inherent srgb->linear and back transforms in wgpu could definitely be useful.  Beware that almost no camera actually records in sRGB or rec709 - nearly every single camera in existence performs an S-curve transform on linear data before applying the sRGB or rec709 transform.  https://www.dpreview.com/forums/post/65804170 shows some of Sony's behaviors. Plus there's the whole wonderful world of logarithmic transform functions.  But scaling S-curve-transformed linear is probably still better than sRGB.  That is, even, if performing DC on linear vs. gamma-compressed data is even the root cause here.

xtrul, 2022-02-17:
i will contact you on the discord server

Entropy512, 2022-02-17:
As a warning, while I am logged in to the Discord server, I'm almost never paying attention there.

AdrianEddy, 2022-02-18:
@Entropy512 I tested Lanczos4 instead of Bilinear and it did give us a quality improvement. However, there's something else going on, check this out: 

https://user-images.githubusercontent.com/3849067/154694599-d5546a74-5dac-490f-9f97-2a0e5817f337.mp4

Entropy512, 2022-02-18:
I would expect Lanczos to have a quality improvement, but I also suspect (like you did) that the root cause of this report lies elsewhere since both the app and OFX are running through the same code path.

As far as the image shifting between "original" and bilinear/lanczos - I'm guessing that's the distortion/FOV functions kicking in?

AdrianEddy, 2022-02-18:
https://user-images.githubusercontent.com/3849067/154739539-7fff55e6-ac42-4bbe-a5cb-07674b666e56.mp4

Still WIP and that's still not in linear colorspace

AdrianEddy, 2022-02-19:
Ok the quality issue is fixed by 267be05ccb7e7ce0671a4a282920bf6dc48728d4 and abb0fd509c6978a16d366f5cad555cfa2c843678

I'll leave this open because there's still one minor issue when using formats with uneven plane dimensions (4:2:0, 4:2:2 but not 4:4:4 or RGB)

Entropy512, 2022-02-19:
I was just about to comment that chroma-subsampled YUV vs RGB would be another possible difference between using the OFX plugin and the app, looks like you may have already found a potential issue there?

While it isn't definitive that the OFX plugin is working on linear data, it is definitely working on RGB and not YUV, subsampled or not.

AdrianEddy, 2022-02-19:
Ok that's fixed too 45a2caef867f43ae440cc4dcc6c0c05f806c4047

closed 2022-02-19 by AdrianEddy

Entropy512, 2022-02-19:
Hmm...  But your offset is 0 for the bilinear case which still doesn't explain the OFX vs app delta.

Although I also wonder - could it just be the results of roundtripping a Resolve export into the app and then back out?

AdrianEddy, 2022-02-20:
@Entropy512 ok so there were two parts to this issue here:
- Bilinear interpolation in the undistortion process
- Fast bilinear interpolation in ffmpeg during pixel format conversion

The first one wasn't that significant here so after implementing it, the result was just slightly better. The real quality killer here was ffmpeg's `sws_scale` function when doing pixel format conversion (eg from YUV420p to YUV422p10le for ProRes).
Default ffmpeg's flag for pixel format conversion was `FAST_BILINEAR` and this makes sense, because it didn't do any *scaling* in there, input and output dimensions were exactly the same, so it just needed to move pixels around between different YUV formats. 

It turned out however, that for some reason this function **is** indeed interpolating the video, even in the Y plane! I don't know why, but just changing this flag to `LANCZOS` made a significant difference in the output quality. I understand there is chroma planes *scaling* happening during conversion between different YUV representations, but in all cases Y plane should remain exactly the same, and yet - it was interpolated too.

I imagine this happens because of insane amount of different pixel formats that ffmpeg supports and ability to convert between any combination of them, so the implementation has to be generic to some degree. 

It could probably help if we covered most common cases on our own, making sure Y plane always stays intact, but since changing this to `LANCZOS` works well, and the amount of different cases we would need to cover is still high - I don't think it's necessary for now.

Third issue appeared after implementing Bicubic and Lanczos4 in undistortion kernels, and it was a slight shift between the planes - and this one was fixed last in 45a2caef867f43ae440cc4dcc6c0c05f806c4047

Entropy512, 2022-02-20:
Oh, that ffmpeg behavior is WEIRD.  It would also perfectly explain why this didn't affect OFX.