gyroflow/gyroflow #180: Gyroflow H264 Hero10 export drops frames in FinalCutPro M1 Air

B-nutze-RR, 2022-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

RC3

### What operating system are you using?

Mac OS

### What GPU are you using?

M1 Air 

### What happened?

H264 95 Mbit/s 59.94 Gopro Hero 10 video exported from Gyroflow skips frames in Finalcut Pro playback .A RSgo export or the source Gopro file play fine in Finalcut Rro .
Attached video shows the issue in screen recording .

https://www.youtube.com/watch?v=9XNnV67FH8Y

reported from @BMSweb on discord.

[image]

### Relevant log output

```shell
not provided yet
```

AdrianEddy, 2022-02-15:
Should be fixed now in https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-02-15:
Fixed in 55e1fb8de7c698f5ede9d4c5dc31f8362a54b501

closed 2022-02-15 by AdrianEddy