gyroflow/gyroflow #181: MacBook Pro M1 perfomance drop

nogod4me, 2022-02-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0-rc3

### What operating system are you using?

macOS Monterey 12.2

### What GPU are you using?

Apple Silicon M1

### What happened?

Testing GF on my GoPro 5 Session with 2.7K 4:3 25FPS footage. After pressing Export first 7-10% of process runs pretty fast, then performance gradually slows till 1 frame per second. GPU encoding goes the same. I have to say that I use my own calibration preset for 2.7K 4by3 on Session 5. 
By the way GoPro Hero10 footage encodes amazingly fast. 
[image]

https://user-images.githubusercontent.com/99600621/153757226-5de5c9d4-887c-41a1-be40-b8368b82c93f.mov

nogod4me, 2022-02-13:
[image]
I can't even finish one video.

AdrianEddy, 2022-02-13:
So it happens only with Hero 5 video? Can you upload it?

Volus208, 2022-02-13:
I can confirm this also happens for me, Ubuntu 20.04 LTS with a Gopro Hero5 Session, and I tried every exporting setting. However I noticed that I can't use my GPU for rendering

nogod4me, 2022-02-13:
[image]

nogod4me, 2022-02-13:
> 

Yes, Session 5 footage runs like this.

nogod4me, 2022-02-13:
> I can confirm this also happens for me, Ubuntu 20.04 LTS with a Gopro Hero5 Session, and I tried every exporting setting. However I noticed that I can't use my GPU for rendering

CPU encoding works well?

HenningSE, 2022-02-14:
I got the same problem with Gopro Hero5 Black footage on an Intel MacBook Pro.

AdrianEddy, 2022-02-14:
This should be fixed in latest dev build https://github.com/gyroflow/gyroflow/suites/5296956914/artifacts/164310984

AdrianEddy, 2022-02-14:
sorry apparently github failed to include it, here's fixed version: https://github.com/gyroflow/gyroflow/suites/5298111892/artifacts/164378495

Volus208, 2022-02-14:
This dev build doesn't work on ubuntu, It starts exporting then says ffmpeg::Error(1668179713: Input changed)
I'm using an intel core i5 9400F and an Nvidia gtx 1050ti

Volus208, 2022-02-14:
It seems to work after the third try, I think the beggining of the file was "corrupted", which happens often on the session 5

HenningSE, 2022-02-14:
The fix is working for me. Thanks for you awesome work.

AdrianEddy, 2022-02-14:
> This dev build doesn't work on ubuntu, It starts exporting then says ffmpeg::Error(1668179713: Input changed) I'm using an intel core i5 9400F and an Nvidia gtx 1050ti

can you send me a file which causes that error?

AdrianEddy, 2022-02-15:
@Volus208 does it work with this version? https://gyroflow.xyz/devbuild/?autodownload

nogod4me, 2022-02-15:
> @Volus208 does it work with this version? https://gyroflow.xyz/devbuild/?autodownload

For me now it works fast but there is such error at the end.. Result file is not playable. 
[image]

AdrianEddy, 2022-02-15:
fixed with latest dev build (gh274) https://gyroflow.xyz/devbuild/?autodownload

nogod4me, 2022-02-15:
> fixed with latest dev build (gh274) https://gyroflow.xyz/devbuild/?autodownload

I just installed it from that link. Then I cut couple frames from the start and the end of file and it worked well.

AdrianEddy, 2022-02-15:
Yes

closed 2022-02-15 by AdrianEddy