gyroflow/gyroflow #186: Unusable after RC2 on MacPro late 2013

Fabersa, 2022-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0 rc3

### What operating system are you using?

MacOS 12.1

### What GPU are you using?

AMD FirePro D700 6gb

### What happened?

I'm trying to get rid of the wobbly video output achieved after 1:15 hours rendering of the RC1 but I've found video preview freeze in RC2 and the impossibility to render on the RC3. I'm forced to use the RC1 but I can't have any good results. The rendering on RC3 failed cause an HWUnknown issue on x264 codec instead I've set the x265 output.... Please help me!

AdrianEddy, 2022-02-14:
what's the exact error message? did you try unchecking "Gpu decoding" under Advanced? what camera is your video file from?

Fabersa, 2022-02-14:
[image]

That's the error window after the GPU rendering, unchecking the GPU rendering it works (same RT) now I'm waiting for the results. The previews had lot of frame dropping and I can't judge if I've reached a good stabilization since for what I can see on the preview my video is still wobbly.

The origin file is from an Insta360 One-r 1"

AdrianEddy, 2022-02-15:
Does it work in this build? https://gyroflow.xyz/devbuild/?autodownload
Can you confirm rc2 works, but rc3 doesn't?

Fabersa, 2022-02-16:
GPU encoding no longer working after RC1

AdrianEddy, 2022-02-16:
What's the error message?

Fabersa, 2022-02-16:
as you can see on my previous attachment : An error occurred: FFmpeg error 542398533 Generic error in an external library
 etc

AdrianEddy, 2022-02-22:
@Fabersa does it work after you disable "GPU decoding" under Advanced?
RC1 didn't support GPU encoding on mac at all - it was using CPU encoder (without showing gpu error)

Fabersa, 2022-02-22:
Yes it works w/o GPU Encoding but it takes a life to encode a minute long clip....

AdrianEddy, 2022-04-28:
Does this still happen in latest dev build?  https://gyroflow.xyz/devbuild/?autodownload

Fabersa, 2022-04-29:
Now seems working again!

closed 2022-04-29 by AdrianEddy