gyroflow/gyroflow #191: GPU encoding not working on RC3 revert to RC2

grabejud, 2022-02-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

RC2

### What operating system are you using?

w10

### What GPU are you using?

gtx 1060 max q

### What happened?

very excited with the new rc3 release, looks good on preview but then upon encoding it gives me 11hrs est, so i know this is not right. Drivers are updated. no choice but revert to rc2 and ran smoothly and all rendered in 2 mins my 3min video.

kwkwkwkw, 2022-02-14:
> Should be fixed already in latest dev build https://gyroflow.xyz/devbuild/?autodownload Please check and let me know

This Dev build having the previous version's (RC1 & RC2) DJI footage export issue (auto shutdown the program) which is fixed in the official RC3 release.

AdrianEddy, 2022-02-15:
Should be fixed now in https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-02-15:
Fixed in 8d38a61c41f8556e7722271f252cc5825a401dd7 and 55e1fb8de7c698f5ede9d4c5dc31f8362a54b501

closed 2022-02-15 by AdrianEddy

grabejud, 2022-02-20:
> Should be fixed now in https://gyroflow.xyz/devbuild/?autodownload

thanks a lot. this one works for me. and the stabilization is much better in this release