gyroflow/gyroflow #194: Add transparency as background option

Janoz-NL, 2022-02-15:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

It would be really nice to have a transparency option for the background. This probably isn't possible for x264 and x265 output, but if it would only work for PNG's it would be more than enough (although I read somewhere ProRes also has transparency options)

relaxgit, 2022-02-15:
set a weird color as background, and chroma key in your editor

Entropy512, 2022-02-15:
ProRes 4444 does support alpha, 422 does not.

AdrianEddy, 2022-02-18:
Done in abb0fd509c6978a16d366f5cad555cfa2c843678
Use hex color code with alpha: `#AARRGGBB`, eg. `#00000000` or just `transparent`

Supported in PNG and ProRes 4444/4444XQ
Latest dev build: https://gyroflow.xyz/devbuild/?autodownload

closed 2022-02-18 by AdrianEddy