gyroflow/gyroflow #195: ProRes export doesn't save video, in/out points don't work - Linux

ShagrathBG, 2022-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0-rc3

### What operating system are you using?

EndeavourOS (arch), Gnome on X11

### What GPU are you using?

Nvidia gtx970

### What happened?

h.265 works, but when i select ProRes export, (proxy or hq, haven't tried the other presets) it looks like it's rendering the frames, but it produces few megabytes .mov file, that only contains sound. Also continues to render frames outside the selected region (i suspect it's from IN to the end of the file)

[image]

### Relevant log output

```shell
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
11:27:24 [INFO] Lens profiles directory: "/home/andrew/Gyroflow/camera_presets"
11:27:24 [INFO] Loaded 1291 lens profiles in 7.694ms
11:27:24 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 970
11:27:24 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
11:27:24 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src/rendering/mod.rs:45] gpu_type = NVIDIA
11:27:24 [INFO] Latest version: v1.0.0-rc3, current version: 1.0.0-rc3 (gh253)
11:27:43 [DEBUG] (6) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
11:27:43 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
11:27:43 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
11:27:43 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
11:27:43 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "thread.video" ""
11:27:43 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.VDPAU"
11:27:43 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
11:27:43 [DEBUG] (9) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7f04dd9c0f50) QSize(873, 655)
get Context 0x7f04de649ae0/0x7f04dc005e88 has no local data for key 0x2, create 1(0x7f04dd855620)
get Context 0x7f04de649ae0/0x7f04dc005e88 has no local data for key 0x1, create 1(0x7f04de56df00)
11:27:43 [DEBUG] (9) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
11:27:43 [INFO] surface size: 959x719
11:27:43 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
11:27:43 [DEBUG] (9) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7f04dc3316d0) QSize(959, 719)
11:28:21 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["vdpau", "cuda", "vaapi", "qsv"]
11:28:21 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_CUDA 0
11:28:21 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_CUDA (cuda) with format Some(AV_PIX_FMT_CUDA)

11:28:21 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("cuda")
11:28:21 [DEBUG] (2) gyroflow::rendering: Possible encoders with NVIDIA: [("prores_ks", false)]
11:28:21 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("prores_ks")
11:28:21 [DEBUG] (2) gyroflow::rendering: start_us: 209905905, render_duration: 26697.09449929479, render_frame_count: 800
[hevc @ 0x7f051c002b40] Hardware is lacking required capabilities
[hevc @ 0x7f051c002b40] Failed setup for format cuda: hwaccel initialisation returned error.
11:28:21 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: None, encoder_pixel_format: Some(YUV422P10LE)
11:28:21 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV422P10LE, color range: JPEG
11:28:21 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type None
11:28:24 [DEBUG] (2) gyroflow::rendering: process_frame: 1, timestamp_us: 209909700
converting to YUV422P10LE
11:28:24 [DEBUG] (2) gyroflow::rendering: process_frame: 2, timestamp_us: 209943067
converting to YUV422P10LE
11:28:24 [DEBUG] (2) gyroflow::rendering: process_frame: 3, timestamp_us: 209976433
converting to YUV422P10LE
11:28:24 [DEBUG] (2) gyroflow::rendering: process_frame: 4, timestamp_us: 210009800
converting to YUV422P10LE
11:28:24 [DEBUG] (2) gyroflow::rendering: process_frame: 5, timestamp_us: 210043167
converting to YUV422P10LE
11:28:25 [DEBUG] (2) gyroflow::rendering: process_frame: 6, timestamp_us: 210076533
converting to YUV422P10LE
11:28:25 [DEBUG] (2) gyroflow::rendering: process_frame: 7, timestamp_us: 210109900
converting to YUV422P10LE
11:28:25 [DEBUG] (2) gyroflow::rendering: process_frame: 8, timestamp_us: 210143267
converting to YUV422P10LE
11:28:25 [DEBUG] (2) gyroflow::rendering: process_frame: 9, timestamp_us: 210176633
converting to YUV422P10LE
11:28:25 [DEBUG] (2) gyroflow::rendering: process_frame: 10, timestamp_us: 210210000
converting to YUV422P10LE

#etc, all the other frames

11:30:05 [DEBUG] (2) gyroflow::rendering: process_frame: 800, timestamp_us: 236569667
converting to YUV422P10LE
```

AdrianEddy, 2022-02-15:
Does it work when you disable "GPU decoding" under Advanced?
`Hardware is lacking required capabilities` this suggest that CUDA decoding failed to initialize. Are you able to run autosync with gpu decoding enabled?

ShagrathBG, 2022-02-15:
When I disable gpu decoding, it doesn't spit the "hardware lacking" line, but the result is the same - it looks like it's rendering from the IN point to the end of the video, but doesn't save any frame in the stabilized file, only audio. Honestly haven't even touched the sync section, since beside the prores export, everything else seems to just work for me, no problems exporting h.265 with gpu for instance.

AdrianEddy, 2022-02-15:
Please try with latest dev build https://gyroflow.xyz/devbuild/?autodownload

ShagrathBG, 2022-02-15:
Works both with gpu decode disabled or enabled, thanks!

closed 2022-02-15 by ShagrathBG