gyroflow/gyroflow #198: Long rendering times, laggy preview

JonaFvp, 2022-02-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0-rc3

### What operating system are you using?

macOS Big Sur 11.6 (20G165)

### What GPU are you using?

Intel HD Graphics 6000 1536 MB

### What happened?

I'm not sure if my mac is just too slow or its the system but i always have laggy preview of the video and exporting 3min files takes like 2 hours...

AdrianEddy, 2022-02-15:
Use latest dev build https://gyroflow.xyz/devbuild/?autodownload and let us know if it's better

kwkwkwkw, 2022-02-17:
> Use latest dev build https://gyroflow.xyz/devbuild/?autodownload and let us know if it's better. Also make sure you have GPU encoding enabled

Tried on this Dev Build GH289 on two windows 10 computer and the playback function seems has problem when playing back the footage.  There is only audio track playback but the footage will stay in the first frame without any response and the timeline indicator doesn't move during the playback.

kwkwkwkw, 2022-02-17:
> Tried on this Dev Build GH289 on two windows 10 computer and the playback function seems has problem when playing back the footage. There is only audio track playback but the footage will stay in [...]

Update : After blind playing back the whole footage for once and re-play it again from the begining, it will then working with both the video and audio fine.

AdrianEddy, 2022-02-17:
Please try again now in latest dev build: https://gyroflow.xyz/devbuild/?autodownload

relaxgit, 2022-02-18:
What you mean laggy preview?
By me i always see a dropped frame on preview at every seconds while playback my footage.
Very disturbing to my eyes :(
CPU and GPU usage is around 25%, Video is 1080 h264 and come from fast ssd. 
In fact it stutters too when i don't have any stabilization at all.

Latest dev build, w10, 16gb ram, 1660ti.
Please fix this issue.

AdrianEddy, 2022-02-19:
@relaxgit Do you have latest NVIDIA drivers? does it happen with all `Preview resolution`s? Does it also happen with `Experimental zero-copy GPU preview` enabled?

fral8, 2022-02-19:
For me on version gh403 slowness problems are solved. 
Windows 10
I7 6700hq
Nvidia GTX 960 m
16 GB  RAM

RD5500, 2022-02-20:
gh306 fixed it for me
Ryzen 7 4800H
GTX1650Ti
8GB DDR4
Windows 10

relaxgit, 2022-02-21:
> @relaxgit Do you have latest NVIDIA drivers? does it happen with all `Preview resolution`s? Does it also happen with `Experimental zero-copy GPU preview` enabled?

I had 49x.x nvidia driver, but now installed latest one, and still have the stuttering bug.
Yes, it happen in all preview resolution and i tried zero-copy method too.

**Please check** my short sample file from here:
https://drive.google.com/file/d/1EEEkVYMakQ2we2myKYhDVhUMwSjJ6zv2/view?usp=sharing

In mediaplayer it's fine, in latest Gyroflow it drop frame in every second.
Many thanks!

AdrianEddy, 2022-02-21:
Is this `drop.mp4` file rendered from gyroflow?

relaxgit, 2022-02-21:
No, its original from a Firefly 8S camera.
It stutters when i don't load any gyro data too..
Just for checking. It stutters by you when you play in gyroflow?

AdrianEddy, 2022-02-22:
> Just for checking. It stutters by you when you play in gyroflow?

Yes, I can reproduce the issue, I'll try to fix it.
It's the only file I've seen this issue with though, all others play fine for me

relaxgit, 2022-02-22:
Thanks for confirm!
Hope it can be fixed.

FranzAr, 2022-02-22:
I have the same issue with runcam thumb video. Tried rc4, all possible settings. Latest Nvidia drivers installed.
Gtx960m, windows 10. Intel i7, 16 GB ram
Sample files https://drive.google.com/file/d/1Zk4XQKFb88NFcTXju85zGGGSyS-pS77Z/view?usp=sharing

AdrianEddy, 2022-03-07:
This should be fixed in latest dev build, please check https://gyroflow.xyz/devbuild/?autodownload

closed 2022-04-30 by AdrianEddy