gyroflow/gyroflow #199: Font sizing too small on 4K monitor

rastarr, 2022-02-16:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Great app, thank you so much!
I'm using a 4K monitor on macOS
Is it possible to add some font scaling to GyroFlow? I'm already wearing glasses however the fonts used throughout (at least for me) are way too small.

relaxgit, 2022-02-16:
there is UI scaling, at leasr in dev mode

rastarr, 2022-02-16:
> there is UI scaling, at leasr in dev mode

I'm using v1-rc3
I don't see any dev mode available to me

Janoz-NL, 2022-02-16:
Its not dev mode, but the latest development build.  ( https://gyroflow.xyz/devbuild/ )

Under 'advanced' on the right there is an UI scaling option.

AdrianEddy, 2022-02-16:
Already done, download latest dev build https://gyroflow.xyz/devbuild/?autodownload  or wait for rc4

closed 2022-02-16 by AdrianEddy