gyroflow/gyroflow #207: Remove duplicated frame  - Runcam Orange 5

NomadaFpv, 2022-02-17:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Add check and remove  duplicated frame grom clip that Runcam Orange 5 used to create.

mycosd, 2022-02-17:
That's a very niche feature. I don't think a stabilizer software should be used for that. You can do the same thing eg. with the `mpdecimate` filter of `ffmpeg`

ElvinC, 2022-02-17:
Yeah... Instructions here: https://docs.gyroflow.xyz/cams/runcam
This is something that should be fixed on the firmware side, which I've already talked to Runcam about. At least in one of the previous firmware versions I tested, it stemmed from a mismatched framerate configuration, though not sure if that's still the case for the latest firmware

closed 2022-02-17 by ElvinC