gyroflow/gyroflow #211: Some nice features

relaxgit, 2022-02-18:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Ok, my list:
- zooming with the mouse wheel is fine, but it should move that part to the **center** of the timeline when i zoom in. Now if i zoom on the right part of the timeline it will move out of the visible area and i have to scroll. Have a look at Magix Vegas, soo clever :)
- invert mouse wheel while scrolling the timeline. If i use down direction scroll the timeline to the **left**
- enable **full screen** preview window while double clicking on it, so we can see our stabilized footage in full glory before rendering
- **scrub** the video when click+drag on the preview window
- show rendering **progress** at the taskbar button, so we can always see the progress while using another app
- display **preview frames** while rendering (0.3 worked like this), and show rendering fps value
- add a **hotkey** for enable/disable stabilizing. Now if i click on the button with mouse, i cannot play the footage with spacebar because lost focus, i have to click back on the timeline
- when i change output size, i have to click on render button **twice** to start rendering
- if i click on advanced panel and it was on the bottom part of the screen it wont be visible, i have to scroll down. Made it **automatically** to show
- is it possible to detect **IMU orientation**?
- 
Thank you devs!!

AdrianEddy, 2022-02-18:
> zooming with the mouse wheel is fine, but it should move that part to the center of the timeline when i zoom in.

This is by design, if I want to take a closer look at a sync point that's near the end of timeline, I just place the mouse there and scroll. With your proposal I would need to scroll in the center and then pan all the way to the right

> display preview frames while rendering (0.3 worked like this)

this will not happen. While rendering there is no RGB image to show, everything happens on separate YUV data, which is not suitable for displaying. Displaying it would double the amount of computations for render, which is pointless

> is it possible to detect IMU orientation?

Tracked in #38 

All others are fine and can be implmented

relaxgit, 2022-02-21:
"With your proposal I would need to scroll in the center"
No, you don't have to scrool to the center.
But i thinked we zoom on that part where the vertical line, the cursor is.
Now i see that this is not the case, we zoom where the mouse cursor is, the vertical line doesn't matter.
This is new to me. In Vegas and other editing apps we click then zoom, and that part will come closer and closer to the center, so we can always see it, and no need for scrolling first to the center. Ok, not a big deal, just a little strange to me :)

Oh, i didn't know that showing yuv data required so much computation.  Also, not a very big deal.

"All others are fine and can be implmented"

Cool, please consider :)

AdrianEddy, 2022-06-10:
> zooming with the mouse wheel is fine, but it should move that part to the center of the timeline when i zoom in. Now if i zoom on the right part of the timeline it will move out of the visible area and i have to scroll. Have a look at Magix Vegas, soo clever :)

It scrolls to the center of playhead there. In Gyroflow it scrolls where your mouse is, independent of the playhead. I prefer how it works here, so it'll stay as is

> invert mouse wheel while scrolling the timeline. If i use down direction scroll the timeline to the left

Already done some time ago

> enable full screen preview window while double clicking on it, so we can see our stabilized footage in full glory before rendering

Done in 50daedda2e8e54b740f2525b5697f165a74bf0e0

> scrub the video when click+drag on the preview window

I'll skip this one for now

> show rendering progress at the taskbar button, so we can always see the progress while using another app

Already done some time ago

> add a hotkey for enable/disable stabilizing. Now if i click on the button with mouse, i cannot play the footage with spacebar because lost focus, i have to click back on the timeline

Already done some time ago, use `s` key

> when i change output size, i have to click on render button twice to start rendering

Fixed in 50daedda2e8e54b740f2525b5697f165a74bf0e0

> if i click on advanced panel and it was on the bottom part of the screen it wont be visible, i have to scroll down. Made it automatically to show

Fixed in 50daedda2e8e54b740f2525b5697f165a74bf0e0

closed 2022-06-10 by AdrianEddy