gyroflow/gyroflow #212: Fedora 35 libc++ issue.

rickiswolf, 2022-02-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0 RC3

### What operating system are you using?

Fedora 35

### What GPU are you using?

Intel iRIS Xe (11th Gen Core i5)

### What happened?

The appimage was not able to start. 

When I tried to use the tar.gz file the following issue was found: 

Terminal: 

[rick@xyz Gyroflow]$ ./gyroflow
./gyroflow: error while loading shared libraries: libc++.so.1: cannot open shared object file: No such file or directory

The packages from the documentation was installed. 

### Relevant log output

```shell
[rick@xyz Gyroflow]$ ./gyroflow
./gyroflow: error while loading shared libraries: libc++.so.1: cannot open shared object file: No such file or directory
```

AdrianEddy, 2022-02-18:
This looks like missing libc++, try with `dnf install libc++ libc++1` or `dnf install libcxx`

rickiswolf, 2022-02-18:
The AppImage is still broken but now the other version is able to start. 

Thank you for the tipp. 

Missing package from my list: 

- libcxx

closed 2022-02-18 by AdrianEddy