gyroflow/gyroflow #213: Multiple .bbl Files for one video

d3dx9, 2022-02-18:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Hi!

Thanks for that nice application. I wonder, is it already possible to combine multiple Betaflight logs? I stop logging as soon I disarm, and this would be really useful :)

Best,
Julian

d3dx9, 2022-03-06:
Hi Elvin,

sorry for the late response, completly oversaw it.

I sadly got only 16MB of internal blackbox storage. I currently got a OsmoAction and it makes it pretty hard to stabilize it with that data. You might have any other idea how I could overcome this issues?

realThirdpartycookie, 2022-03-16:
Yes, with Betaflight 4.3 you can disable all the columns that are unnecessary for gyroflow,
like that you save up huge amounts of space. additionally you can reduce the logging rate.

AdrianEddy, 2022-07-29:
You can just append the files together and it should be working fine, on windows it would be something like `COPY /B log1.txt + log2.txt log_out.txt`
on linux/mac, it would be `cat log1.txt log2.txt > log_out.txt`

closed 2022-07-29 by AdrianEddy