gyroflow/gyroflow #214: a7c xavc s 4k gyro video Crashes

justdoyour, 2022-02-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0 rc 3 ,2,1

### What operating system are you using?

Windows 11

### What GPU are you using?

Nvidia 1050

### What happened?

My Sony a7c 4k 30fps xavc s video crashes everytime I try to load it. If I load a mp4 with the same characteristics but no gyro data it does fine. But otherwise it crashes.

### Relevant log output

```shell
PS C:\Users\justd\videos\stabilize> .\gyroflow.exe
PS C:\Users\justd\videos\stabilize> 15:52:50 [INFO] Latest version: v1.0.0-rc3, current version: 1.0.0-rc3
15:52:50 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\justd\\Videos\\Stabilize\\camera_presets"
15:52:50 [INFO] Loaded 1294 lens profiles in 34.549ms
15:52:50 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 1050
15:52:51 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
15:52:51 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src\rendering\mod.rs:45] gpu_type = NVIDIA
15:53:01 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
15:53:01 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.video" ""
15:53:01 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
15:53:01 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
15:53:01 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
15:53:01 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
15:53:01 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x239c499f560) QSize(1760, 990)
15:53:02 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.video" "MFT"
15:53:02 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
15:53:02 [INFO] surface size: 1920x1080
15:53:02 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
15:53:02 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x239e1fcf4e0) QSize(1920, 1080)
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: Error { kind: UnexpectedEof, message: "failed to fill whole buffer" }', C:\Users\runneradmin\.cargo\git\checkouts\telemetry-parser-19821e7ac6d09f51\665f9b0\src\tags_impl.rs:209:72
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Rayon: detected unexpected panic; aborting
```

justdoyour, 2022-02-18:
> Ok, should be a simple issue, can you send me that file?

the video file?

justdoyour, 2022-02-18:
its 2gb. I'm uploading now

AdrianEddy, 2022-02-18:
Fixed in https://github.com/AdrianEddy/telemetry-parser/commit/d14cf06a04ab7a0770fb0c51d3545dfe304f32e9
Latest dev build: https://gyroflow.xyz/devbuild/?autodownload

closed 2022-02-18 by AdrianEddy