gyroflow/gyroflow #221: Frame readout time in saved file is overriden by lens profile.

Entropy512, 2022-02-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1bcb76c2807e9dabf900eca3604836e525a23949

### What operating system are you using?

Ubuntu 21.10

### What GPU are you using?

NVidia RTX 3070

### What happened?

If using a lens profile for a camera that has the same lens configuration and crop factor but a different readout rate (such as A7M4+FE85 using the A7S3+FE85 combo), any changes saved in the ,gyroflow file are not read back - the lens profile readout rate takes priority.

Steps to reproduce:
Set up a project with a lens profile that specifies readout rate, such as FE85+A7S3
Change the frame readout time slider
Save a .gyroflow file for this project
Reload the .gyroflow file and note that any changes made by the user to that slider are undone

Entropy512, 2022-02-21:
I have confirmed that the configured readout time is set in the .gyroflow file:
```
    "frame_readout_time": 26.8,
    "adaptive_zoom_window": 0,
    "adaptive_zoom_fovs": []
```
However the frame readout time is also specified in the lens profile section.

The value outside of the lens profile should take precedence when loading a .gyroflow file

AdrianEddy, 2022-02-21:
Should be fixed in 3b86dd6ca337a7e8510674d60f498437bf443482

closed 2022-02-21 by AdrianEddy