gyroflow/gyroflow #222: Regression:  Video is constantly overlaid with a "waiting" spinner

Entropy512, 2022-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

3b86dd6ca337a7e8510674d60f498437bf443482

### What operating system are you using?

Ubuntu 21.10

### What GPU are you using?

NVidia RTX 3070

### What happened?

Starting with commit 3b86dd6ca337a7e8510674d60f498437bf443482 - the video preview window is now overlaid constantly with a spinning "loading" circle once a video is loaded.

(Side note, you'll see a crash/coredump in the log, I'll file a bug report on this tomorrow with a backtrace.  It hasn't been a high priority for me as it happens when exiting the program intentionally.)

### Relevant log output

```shell
Compiling gyroflow v1.0.0-rc3 (/home/adodd/gitrepos/gyroflow)
   Compiling gyroflow-core v1.0.0-rc3 (/home/adodd/gitrepos/gyroflow/src/core)
    Finished release [optimized] target(s) in 26.72s
     Running `target/release/gyroflow`
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
00:15:57 [INFO] Lens profiles directory: "/home/adodd/gitrepos/gyroflow/resources/camera_presets"
00:15:57 [INFO] Loaded 1292 lens profiles in 7.099ms
00:15:57 [INFO] Latest version: v1.0.0-rc3, current version: 1.0.0-rc3 (dev4953)
00:15:58 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce RTX 3070 Laptop GPU
00:15:58 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
00:15:58 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src/rendering/mod.rs:47] gpu_type = NVIDIA
00:15:59 [DEBUG] (6) gyroflow::util: Minidump uploaded: OK
00:16:07 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
00:16:07 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.video" ""
Failed to open VDPAU backend libvdpau_va_gl.so: cannot open shared object file: No such file or directory
Failed to open VDPAU backend libvdpau_va_gl.so: cannot open shared object file: No such file or directory
00:16:07 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
00:16:07 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
00:16:07 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
00:16:07 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.MMAL.mmal"
00:16:07 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
00:16:07 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fcb8011f490) QSize(1031, 580)
get Context 0x7fcb805ba250/0x7fcb80007170 has no local data for key 0x2, create 1(0x7fcb80675160)
get Context 0x7fcb805ba250/0x7fcb80007170 has no local data for key 0x1, create 1(0x7fcb80196fb0)
00:16:08 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
00:16:08 [INFO] surface size: 1280x720
00:16:09 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
00:16:09 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fcb8170ac40) QSize(1280, 720)
thread '<unnamed>' panicked at 'use of std::thread::current() is not possible after the thread's local data has been destroyed', library/std/src/thread/mod.rs:656:35
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
fatal runtime error: Rust panics must be rethrown
Crashdump written to /home/adodd/gitrepos/gyroflow/46c949ee-06ef-439c-bb750997-cd821407.dmp
Aborted (core dumped)
```

AdrianEddy, 2022-02-22:
hmm, I can't reproduce on my end, does it happen with all videos? does the loader go away after you move any smoothing sliders?

AdrianEddy, 2022-02-22:
If anything it would be this change in VideoArea.qml: https://github.com/gyroflow/gyroflow/commit/3b86dd6ca337a7e8510674d60f498437bf443482#diff-3e56677c3ad4daf314196186eab2b5fa55afc66046eb7d593ee0ac3170600fe9L180-R198
Can you try reverting it on your side and confirm?

Entropy512, 2022-02-22:
Hmm.  It appears to only happen if loading a ,gyroflow file, regardless of whether gyro data is present or not.

If I drop a video on the window after loading the change, it is sometimes still affected but not always.  If I load a video first and THEN load the gyroflow file, the problem does not occur.

Issue goes away when moving the smoothness slider.

What is making me really scratch my head is that reverting just that file does not seem to fix the issue...  Even though it was my first guess.

AdrianEddy, 2022-06-10:
Fixed in 7befed46d8a7d73fa6150df634dc8ffcb297ef04

closed 2022-06-10 by AdrianEddy