gyroflow/gyroflow #225: Gyroflow Timecode sync feature

Thomaswb2, 2022-02-22:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

It would be great if there was a "sync timecode" button that synced the unix timed gyro data from android phones and video timecode on the timeline, for use with mobile phones which have been jam synced with cinema cameras.

AdrianEddy, 2024-06-05:
Is this still relevant/needed? can you describe your workflow a bit where would this be helpful? Can you share some sample files?

zorrobyte, 2024-06-22:
> Is this still relevant/needed? can you describe your workflow a bit where would this be helpful? Can you share some sample 

I have a Canon R5 C that has a BNC Timecode out port, I'm building an ESP32/Arduino timecode reader with a BMI270 IMU which should 100% solve the external timecode issue for Timecoded cameras. The camera only generates timecode and is frame by frame accurate.

Something like this is easy to build: https://escmdxi.wordpress.com/2022/08/16/speed-reading-ltc/

I've only begun research as of today, nice that this project is OSS. I've built several ESP32/Arduino projects in the past, so I'm not completely spitballing.

My workflow will be to record then slap both SD cards in my machine. Gyroflow should be able to sync timecode (with a fudge factor if there is any delay).

In the end, just need to add the LTC data to your existing format: https://docs.gyroflow.xyz/app/technical-details/gcsv-format

closed 2025-09-04 by AdrianEddy