gyroflow/gyroflow #226: MacOS Gopro Hero 10 footage becomes unstabilized after modifying settings

AdrianLSY, 2022-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0-RC3

### What operating system are you using?

MacOS 12.2.1

### What GPU are you using?

Macbook air M1 2020

### What happened?

Using GoPro Hero 10 2.7k No-EIS 60fps Wide

After loading the GoPro footage, the application correctly stabilizes the footage. Touching any setting (i.e: FOV, smoothness, lens profile) the footage becomes unstabilized

AdrianEddy, 2022-02-22:
This should be fixed in latest dev build (gh316): https://gyroflow.xyz/devbuild/?autodownload
Please test and confirm

AdrianEddy, 2022-02-22:
Please reopen if this still happens in v1.0.0-rc4

closed 2022-02-22 by AdrianEddy