gyroflow/gyroflow #227: Gyroflow crashes under Linux (AppImage too)

sherghan, 2022-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0-rc4

### What operating system are you using?

Fedora 35 / Gnome / X11

### What GPU are you using?

Intel UHD Graphics 620 / Nvidia GeForce MX250

### What happened?

I'm unable to run any Linux version of the Gyroflow on Fedora (running X11).

When I provide LD_LIBRARY_PATH I am getting UI, but it crashes while opening video file (Caddx Peanut).

Are core dumps interesting? If so, I can attach them.

### Relevant log output

```shell
$ Gyroflow/gyroflow
Gyroflow/gyroflow: error while loading shared libraries: libc++.so.1: cannot open shared object file: No such file or directory
FAIL: 127

$ LD_LIBRARY_PATH=/usr/lib64/lib ./Gyroflow/gyroflow
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
17:19:48 [DEBUG] (2) gyroflow::util: Minidump uploaded: OK
17:19:49 [DEBUG] (2) gyroflow::util: Minidump uploaded: OK
17:19:50 [INFO] Lens profiles directory: "/home/sherghan/Downloads/Gyroflow/camera_presets"
17:19:50 [INFO] Loaded 1293 lens profiles in 8.353ms
17:19:50 [INFO] Latest version: v1.0.0-rc4, current version: 1.0.0-rc4 (gh321)
17:19:50 [INFO] OpenCL Platform: Portable Computing Language, Device: GenuineIntel pthread-Intel(R) Core(TM) i7-8565U CPU @ 1.80GHz
17:19:50 [WARN] Unknown GPU genuineintel pthread-intel(r) core(tm) i7-8565u cpu @ 1.80ghz
[src/rendering/mod.rs:47] gpu_type = Unknown
17:19:57 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
17:19:57 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
17:19:57 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.video" ""
17:19:57 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
17:19:57 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
17:19:57 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
17:19:57 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7f29d426d5e0) QSize(716, 716)
libva info: VA-API version 1.13.0
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_13
libva info: va_openDriver() returns 0
17:19:57 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.VDPAU"
get Context 0x7f29d46ac780/0x7f29d4005de0 has no local data for key 0x2, create 1(0x7f29d4348bb0)
get Context 0x7f29d46ac780/0x7f29d4005de0 has no local data for key 0x1, create 1(0x7f29d4118cd0)
17:19:57 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
17:19:57 [INFO] surface size: 2688x2688
17:19:57 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
17:19:57 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7f29d4dcc650) QSize(2688, 2688)
Crashdump written to /home/sherghan/Downloads/8f9617a1-e483-4902-dcfa6ab5-9186053d.dmp
[1]    214612 segmentation fault (core dumped)  LD_LIBRARY_PATH=/usr/lib64/lib ./Gyroflow/gyroflow
FAIL: 139

$ ./Gyroflow-linux64.AppImage
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
Crashdump written to /home/sherghan/Downloads/7df8936c-d463-4bf9-3b7c359c-f95e1828.dmp
[1]    215199 segmentation fault (core dumped)  ./Gyroflow-linux64.AppImage
FAIL: 139
```

schnupperm, 2022-02-23:
you need the  libc++-dev package.
Please read the release docs:
you need to do:
> sudo apt install libva2 libvdpau1 libasound2 libxkbcommon0 libpulse0 libc++-dev vdpau-va-driver libvulkan1

schnupperm, 2022-02-23:
>Still AppImage should work without a problem.

should!
but on kubuntu 21.10 it crashes also (all -rc versions).

alex116, 2022-02-23:
ArchLinux trying with https://github.com/gyroflow/gyroflow/releases/download/v1.0.0-rc4/Gyroflow-linux64.AppImage
```
Linux xps15 5.16.9-arch1-1 #1 SMP PREEMPT Fri, 11 Feb 2022 22:42:06 +0000 x86_64 GNU/Linux

./Gyroflow-linux64.AppImage 
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
Crashdump written to /home/alex/Downloads/87e00b85-e476-4a4e-941f74bf-0e649591.dmp
[1]    102390 segmentation fault (core dumped)  ./Gyroflow-linux64.AppImage
```
[gyroflow-dump.zip](https://github.com/gyroflow/gyroflow/files/8126122/gyroflow-dump.zip)

mbuzdalov, 2022-02-24:
Well, I have to admit that the Linux version of this "new" gyroflow is too overfitted to Ubuntu and it is completely impossible to run it on an environment that is just a little bit different - e.g. no pulseaudio => screwed, no opencl => screwed, paths are non-Ubuntical => screwed, etc. That's sad, the old Python-based gyroflow at least could be made to work.

adworacz, 2022-03-01:
Same as alex116, I'm crashing on Arch Linux (Manjaro specifically) with the AppImage of RC4.

```
~/Downloads/Gyroflow >>> ./Gyroflow-linux64.AppImage
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
Crashdump written to /home/user/Downloads/Gyroflow/aec3f2fa-7a6e-4bc7-4fd359b0-72141741.dmp
zsh: segmentation fault (core dumped)  ./Gyroflow-linux64.AppImage
```

[gyrodump.zip](https://github.com/gyroflow/gyroflow/files/8157930/gyrodump.zip)

tpetazzoni, 2022-03-05:
I confirm that it crashes under Fedora 35:

```
~/Downloads/Gyroflow-linux64.AppImage --help
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
Crashdump written to /tmp/69329eb4-55c7-458a-3d169fa8-c2f0018d.dmp
Segmentation fault (core dumped)
```

Lord3n20, 2022-03-05:
AppImage isnt working for me either
You can download the .tar.gz or build it from source
This helps to find the error and worked for me.

adworacz, 2022-03-05:
Try the fix I detailed in this other issue:

https://github.com/gyroflow/gyroflow/issues/154

juanjax, 2022-04-22:
Crashing with same issues with MX Linux 21!

AdrianEddy, 2022-11-18:
I believe this is fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload