gyroflow/gyroflow #228: Does not stabilize the video according to INAV magazine

GarryBu, 2022-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1-RC4

### What operating system are you using?

Windows

### What GPU are you using?

NVidea GForce 110

### What happened?

INAV Blackbox Logs (TXT format) are detected and INAV version is shown but no gyro traces appear and no sync possible.

https://user-images.githubusercontent.com/60334441/155193448-9d8b4bbf-5f85-41d0-8153-ea04d5b69cad.mp4

[LOG00007.zip](https://github.com/gyroflow/gyroflow/files/8119149/LOG00007.zip)

### Relevant log output

```shell
18:14:24 [INFO] Lens profiles directory: "\\\\?\\C:\\Gyroflow\\camera_presets"
18:14:24 [INFO] Loaded 1293 lens profiles in 20.179ms
18:14:24 [INFO] Latest version: v1.0.0-rc4, current version: 1.0.0-rc4
18:14:24 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation GeForce MX110
18:14:24 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
18:14:24 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src\rendering\mod.rs:47] gpu_type = NVIDIA
18:14:36 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
18:14:36 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.video" ""
18:14:36 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
18:14:36 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
18:14:36 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
18:14:36 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.video" "MFT"
18:14:36 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
18:14:36 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x2090c1ece30) QSize(1146, 645)
18:14:37 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
18:14:37 [INFO] surface size: 853x479
18:14:37 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
18:14:37 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x2090c1ec380) QSize(853, 479)
18:15:07 [WARN] Sync point out of acceptable range 498.96000000000004 < 450
18:15:07 [WARN] Sync point out of acceptable range 498.97 < 450
18:15:07 [WARN] Sync point out of acceptable range 498.96 < 450
18:15:07 [INFO] Setting offset at 192750.0000: 723.2600 (cost 12650.2804)
18:15:07 [INFO] Setting offset at 321250.0000: 1239.1400 (cost 667476.3815)
```

AdrianEddy, 2022-02-25:
please upload the corresponding video file, 
this INAV log is parsed correctly so probably nothing wrong here

GarryBu, 2022-02-27:
This is the beginning of the video with this log
https://drive.google.com/file/d/1m2MH_yqSyL-JDNcBxJyz9XxmdT2hJe7e/view?usp=sharing 
[LOG00007.zip](https://github.com/gyroflow/gyroflow/files/8148325/LOG00007.zip)

AdrianEddy, 2022-02-27:
are you sure this is the correct log and video file? Movements in these files don't look similar.
Did you have your camera mounted at an angle? what was the rough gyro offset? I don't see any issue with Gyroflow here, you just need to set your parameters correctly and synchronize gyro+video properly

GarryBu, 2022-02-27:
Here the video is cropped, but the log is complete.  This is the very beginning of the video.  The flight controller is rotated 90 degrees.  I figured that all axes should be determined automatically.  But he changed the settings himself, not a single change led to proper stabilization.

GarryBu, 2022-03-02:
Increased the time to search for control points - began to synchronize. It can be closed, it works.

closed 2022-03-02 by GarryBu