gyroflow/gyroflow #229: Why does my GPU isn't actually loaded while encoding through it?

timmal, 2022-02-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0-rc4

### What operating system are you using?

macOS Monterey 12.2.1

### What GPU are you using?

AMD Radeon Pro 5600M

### What happened?

Why does my GPU isn't actually loaded while encoding through it?

[image]

AdrianEddy, 2022-02-22:
Most likely because OpenCL on some Macs (or all Intel ones) uses CPU. It should be fixed when #12 is done

AdrianEddy, 2022-02-24:
Please try latest dev build and let me know if it changed anything in this regard: https://gyroflow.xyz/devbuild/?autodownload
the app title should say `v1.0.0-rc4 (gh341)`

timmal, 2022-03-01:
GPU is works now. But there is no preview and output result have artefacts 

[image]

[image]

AdrianEddy, 2022-03-07:
Fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-03-07 by AdrianEddy