gyroflow/gyroflow #235: Load Presets for Lens Profile

DrewDronesFPV, 2022-02-23:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

- I would like the ability to save a preset for certain lenses.

Ex: If I select a specific Lens Profile (e.g Black Magic 4K Laowa 7.5MM) It would automatically change export settings to ProRes HQ 1000MBps bitrate with Xyz IMU Orientation and any other saved settings one would like included

I am not sure if it should be under a new header "Load Preset" or under Lens Profile but would save time! 

Thanks!

Entropy512, 2022-02-23:
Hmm, would an appropriate way to describe this be to have presets in a .gyroflow file, but NOT specify an actual video clip and gyro data offset?  (e.g. ability to export or load only some settings but not all through a .gyroflow file?)

DrewDronesFPV, 2022-02-23:
> 

I am not 100% sure to understand, but the example you give seems correct. 

Here is my current situation, every time I choose a specific lens profile, I know I will for sure always be changing the IMU orientation because of how my GoPro is mounted on that specific camera + my export settings will always be ProRes 422 HQ.

Entropy512, 2022-02-23:
Yeah - and right now, those are stored in an exported .gyroflow file, but also a bunch of stuff I'm guessing you DON'T want to be touched (video clip and gyro data source, other settings) is not.

Possible solutions might be:

1. Only export some settings in a .gyroflow file - would require modifying the export dialog.  (Right now, there are two settings available, with/without gyro data).
2. Only import some settings from a .gyroflow file when importing/loading

I personally think item 1) is likely better, since you would decide once which settings you did NOT want saved, vs. deciding what you didn't want in every single import with item 2.

I might be able to dig up a similar example of such a UI from another piece of software tonight just as a design idea, but someone else will have to implement it.  I am HORRIBLE at software UI/UX in general, especially implementing GUIs.

DrewDronesFPV, 2022-02-24:
> I personally think item 1) is likely better, since you would decide once which settings you did NOT want saved, vs. deciding what you didn't want in every single import with item 2.

Exactly!

Entropy512, 2022-03-14:
As an FYI for an example of a similar implementation, what I was thinking of are RawTherapee partial profiles:
http://rawpedia.rawtherapee.com/Creating_processing_profiles_for_general_use
http://rawpedia.rawtherapee.com/Sidecar_Files_-_Processing_Profiles#Partial_Processing_Profiles_and_Fill_Modes

Something similar to that dialog could be applied when saving a .gyroflow preset file

AdrianEddy, 2022-06-14:
Exporting presets is now supported in GUI as of 7a4d44e56cb7ea1f13339c46575b6d4d00752384

closed 2022-06-14 by AdrianEddy