gyroflow/gyroflow #239: Rendering FPS Speed

DrewDronesFPV, 2022-02-24:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

One feature that would be cool to implement and would make things easier to see if renders become slower and faster between Release Candidates would be to have a Rendering FPS speed. 

Example: (Rendering 3.1FPS per second)

relaxgit, 2022-02-24:
yes, i also like to see this info.
along with a progress bar at the app button on taskbar. so i can do other things while rendering and know exactly when finished.

AdrianEddy, 2022-02-25:
@DrewDronesFPV done in 7dc09b353c07cd8b0b5457ba23d2b64c362260fd
@relaxgit done in c0e73bbf82eda1a06fba2457b0c9f8c52991dd55

relaxgit, 2022-02-25:
wow, that was fast :)

DrewDronesFPV, 2022-02-25:
that's awesome dude!