gyroflow/gyroflow #241: "Auto Sync" and "Auto Calibrate" get stuck a 2% since RC3 with Sony A7III files

maximecaron, 2022-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

RC2

### What operating system are you using?

mac book pro M1 CPU

### What GPU are you using?

mac book pro M1 CPU

### What happened?

"Auto Sync" and "Auto Calibrate"  was working in rc2 but not working in RC3 and RC4 for Sony A7III video files.

### Relevant log output

```shell
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x10f009c00] st: 0 edit list: 1 Missing key frame while searching for timestamp: 1001
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x10f009c00] st: 0 edit list 1 Cannot find an index entry before timestamp: 1001.
19:53:50 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_VIDEOTOOLBOX 0
19:53:50 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_VIDEOTOOLBOX (videotoolbox) with format Some(AV_PIX_FMT_VIDEOTOOLBOX)

19:53:50 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:50 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:50 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
[h264 @ 0x10ea76200] hardware accelerator failed to decode picture
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
19:53:51 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
```

AdrianEddy, 2022-02-24:
try unchecking "Use GPU decoding" under Advanced and let me know if that worked

maximecaron, 2022-02-24:
in RC4 without "Use GPU decoding" it's working perfectly thanks

AdrianEddy, 2022-03-08:
Fixed in 83e833cf0b47bb1183985faa7f0a89ab7afb9d60

closed 2022-03-08 by AdrianEddy