gyroflow/gyroflow #242: 10bit DCI 4k from the BGH1 - Gyroflow rc4 shows a garbled image

alexrjohnston, 2022-02-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

rc4 and nightly

### What operating system are you using?

Mojave - mackbook pro and hackintosh same issue

### What GPU are you using?

750M and RX580

### What happened?

[image]
[image]

10bit DCI 4k from the BGH1 Garbled image - audio plays as it should. I tried with GPU encoding disabled as per a recommendation from Alexander G. Vedeler
The same file plays properly in gyroflow on my Windows 10 system.

AdrianEddy, 2022-02-24:
can you upload sample file which does that?

alexrjohnston, 2022-02-24:
I will get it uploading now and send the link in the morning!

Regards,
Al.

alexrjohnston, 2022-02-25:
Here is some footage that does it on both my systems: https://drive.google.com/file/d/1zDVLqEV5S_yB74rFA69H_iQh5XxNJNkh/view?usp=sharing

Regards,
Al.

alexrjohnston, 2022-03-10:
Sadly this is still happening on rc5, any ideas?

alexrjohnston, 2022-03-10:
[image]

this is from a GH5s, 10bit. same thing happens with 10 bit bgh1 footage as linked in the file in my previous post. 

macbook pro 2014 and hackintosh with RX580 - same issue

alexrjohnston, 2022-03-10:
If I then open another file after the first then it is just a solid colour which fades between green blue and grey instead of showing me the video as expected. I can drag in gopro footage, iphone footage etc and it plays as expected

[image]

mycosd, 2022-03-11:
Can you try and export the video from GF, so we can check if it's just the preview player messing up. I can't reproduce it on my Win system, the sample video you provided plays fine.

alexrjohnston, 2022-03-11:
Thanks for the reply, yes it works perfectly on windows. I will try an export later and report back.

alexrjohnston, 2022-03-11:
I have just tested an export and it is indeed just the preview which has the issue, the final file is properly rendered

Regards,
Al

mycosd, 2022-03-11:
Under `Advanced` uncheck `Experimental zero-copy preview`. It should be off by default, but in your screenshot it's on. You'll need to hit play afterwards to see a difference, and maybe even restart GF (in case it got stuck somehow)

alexrjohnston, 2022-03-11:
Thanks for the message, it is now off and the issue persists. I had just tried it when I took the screenshot I believe

AdrianEddy, 2022-03-11:
This looks like issue in MDK @wang-bin

alexrjohnston, 2022-03-11:
Is there anything I can do to help push this forwards?

wang-bin, 2022-03-12:
I can't reproduce it on my m1 macbook. I need the log

alexrjohnston, 2022-03-12:
Thanks, how do I provide this?

AdrianEddy, 2022-03-12:
Open package contents -> Contents -> macOS -> drag `gyroflow` file into terminal and press enter. Then after you close the app, upload gyroflow.log file

alexrjohnston, 2022-03-13:
Thanks for the info, here is the log as requested. let me know if you need anything else

Regards,
Al

[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8239852/gyroflow.log)

wang-bin, 2022-03-14:
@AdrianEddy  I see many opencl errors in the log, is it the reason?

AdrianEddy, 2022-03-14:
> @AdrianEddy I see many opencl errors in the log, is it the reason?

Probably not, if OpenCL failed it fallsback to wgpu implementation

AdrianEddy, 2022-03-14:
@alexrjohnston please try latest dev build and upload the log again 
https://gyroflow.xyz/devbuild/?autodownload

alexrjohnston, 2022-03-14:
Thanks for the update, sadly the same issue, log attached
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8247305/gyroflow.log)
[image]

AdrianEddy, 2022-03-14:
How does it look like when you set FOV to 2?

alexrjohnston, 2022-03-14:
If you rotate or zoom out it zooms and rotates the garbled image, it also applies the lens profiles to it correctly

AdrianEddy, 2022-03-14:
@wang-bin please check the log again, no more OpenCL errors

wang-bin, 2022-03-15:
Nothing wrong in the log, except the decoder output format is nv12, on my m1 it's p210, i guess gpu is unable to output 10bit frames, but it should not matter.  More tests will be helpful
- Use VT decoder with memory readback
```
    export MDK_DECODERS="VT:copy=1"
    /path/of/gyroflow
```
- Use VT software decoder
```
    export MDK_DECODERS="VT:hardware=0"
    /path/of/gyroflow
```
- use opengl renderer
```
    export MDK_DECODERS="VT"
    export QSG_RHI_BACKEND=opengl
    /path/of/gyroflow
```

alexrjohnston, 2022-03-15:
How do I action this? pasting these commands into the terminal doesn't seem to be doing the trick. tia

alexrjohnston, 2022-03-15:
Okay that is taking the commands now, I tried all of them at once and I just get a spinning circle now instead of playing garbled footage on the RX580

Do i need to test each set of settings changes on a fresh install of gyro flow or can I just add each change, try and then add the next? ta

alexrjohnston, 2022-03-15:
log on radeon with all setting above applied:
[gyroflow 1.log](https://github.com/gyroflow/gyroflow/files/8256117/gyroflow.1.log)

AdrianEddy, 2022-03-15:
You can test each one on the same installation but use a new terminal window. Upload gyroflow.log after each test

AdrianEddy, 2022-03-15:
That's correct, yes, and these are 3 separate tests

alexrjohnston, 2022-03-15:
@wang-bin I just noticed your question about quicktime - very good point it cannot play these files on mac! you have to use VLC player

does this help? I am just about to do the tests

Regards,
Al.

alexrjohnston, 2022-03-15:
I am copying the following into the terminal and getting the corresponding logs: 

export MDK_DECODERS="VT:copy=1"
/Volumes/750GB/Downloads/Gyroflow-nightly.app/Contents/MacOS/gyroflow
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8257175/gyroflow.log)
>> still garbled

export MDK_DECODERS="VT:hardware=0"
/Volumes/750GB/Downloads/Gyroflow-nightly.app/Contents/MacOS/gyroflow
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8257178/gyroflow.log)
>> spinning wheel instead of garbled

export MDK_DECODERS="VT"
export QSG_RHI_BACKEND=opengl
/Volumes/750GB/Downloads/Gyroflow-nightly.app/Contents/MacOS/gyroflow
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8257185/gyroflow.log)
>> still garbled

wang-bin, 2022-03-16:
> @wang-bin I just noticed your question about quicktime - very good point it cannot play these files on mac! you have to use VLC player

So gpu decoder does not support 422 10bit avc on your machine. The decoder should report an error but it doesn't, it's apple's fault. What i can do is using a blacklist.

VLC does not support h264 10bit hardware decoding on mac,  software decoder is used. You can use software decoder in gyroflow too

```
export MDK_DECODERS="FFmpeg"
/path/of/gyroflow
```

alexrjohnston, 2022-03-16:
<3 that is the problem for sure!

huge thanks all. export MDK_DECODERS="FFmpeg" shows the video preview as expected. I wonder if later versions of the mac OS support 10 bit on intel or if it is only after the jump to M1

Regards,
Al.

AdrianEddy, 2022-03-16:
Upgrading to latest macOS version is a good idea anyway, there is a chance this is fixed there

alexrjohnston, 2022-03-16:
Thanks for all the info @AdrianEddy, so good that it is up and running.

alexrjohnston, 2022-03-16:
Thanks for all your work on this, it's incredible software

Regards,
Al.

AdrianEddy, 2022-05-01:
@alexrjohnston I believe this is fixed in latest mdk-sdk, can you confirm it works in latest dev build of Gyroflow without additional flags?
https://gyroflow.xyz/devbuild/?autodownload

alexrjohnston, 2022-05-04:
Now fixed! Thanks for your hard work all

Regards,
Al.

closed 2022-05-04 by AdrianEddy