gyroflow/gyroflow #245: `wgpu` doesn't work on Intel Mac

AdrianEddy, 2022-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

all

### What operating system are you using?

macOS Intel

### What GPU are you using?

AMD Radeon

### What happened?

for some reason `wgpu` on macOS, even though properly selected AMD Radeon discrete GPU with Metal backend - makes the app unresponsive 

### Relevant log output

```shell
MDK 0.14.0 (git f7623f2) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS1090 x86_64; AppleClang13.0.0 (clang-1300.0.29.30)GCC4.2.1; libc++12000; 12:21:15 Feb 20 2022
05:54:23 [WARN] Populating font family aliases took 232 ms. Replace uses of missing font family "Arial" with one that exists to avoid this cost. 
05:54:24 [INFO] Latest version: v1.0.0-rc4, current version: 1.0.0-rc4 (gh347)
05:54:24 [INFO] Lens profiles directory: "/Applications/Gyroflow.app/Contents/Resources/camera_presets"
05:54:24 [INFO] Loaded 1293 lens profiles in 12.511ms
05:54:24 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Apple") [
    "Intel(R) Core(TM) i9-9980HK CPU @ 2.40GHz",
    "Intel(R) UHD Graphics 630",
    "AMD Radeon Pro 5500M Compute Engine",
]
05:54:24 [INFO] OpenCL Platform: Apple, Device: Intel Intel(R) Core(TM) i9-9980HK CPU @ 2.40GHz
05:54:24 [WARN] Unknown GPU intel intel(r) core(tm) i9-9980hk cpu @ 2.40ghz
[src/rendering/mod.rs:47] gpu_type = Unknown
05:54:32 [DEBUG] (18) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
05:54:32 [DEBUG] (19) MDKPlayer: [in m_player->onEvent] "thread.video" ""
05:54:32 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
05:54:32 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
05:54:32 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
05:54:32 [DEBUG] (19) MDKPlayer: [in m_player->onEvent] "decoder.video" "VideoToolbox"
AudioQueuePause @4437.000 (sample time)/61709481212850 (host time)
05:54:32 [DEBUG] (21) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Metal
05:54:32 [DEBUG] (21) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x600002b218c0) QSize(1309, 982)
05:54:33 [DEBUG] (21) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
05:54:33 [INFO] surface size: 959x719
AudioQueuePause @11108.000 (sample time)/61710269322152 (host time)
05:54:33 [DEBUG] (21) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "AMD Radeon Pro 5500M", vendor: 0, device: 0, device_type: DiscreteGpu, backend: Metal }
AudioQueuePause @17986.000 (sample time)/61710430548188 (host time)
05:54:38 [DEBUG] (21) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Metal
05:54:38 [DEBUG] (21) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x600002a023a0) QSize(959, 719)
AudioQueuePause @24761.000 (sample time)/61722700276889 (host time)
AudioQueuePause @31533.000 (sample time)/61849952087297 (host time)
```

scottyblue77, 2022-02-27:
I i have same config with i9 and radeon gpu vega one VII and i can say the app itself woked quite ok, but the rendering is so slow and not using GPU, all other apps working well with GPU support. So hoping it is something ez to solve. If i boot with windows i am fine and can render in less 5 min 4GB.

AdrianEddy, 2022-02-27:
Did you try latest dev build? https://gyroflow.xyz/devbuild

scottyblue77, 2022-02-27:
Hi, this version shows black screen in preview and give this error if i not change any settings and load hero10 file straight. If i take out the gpu encoding it goes with cpu and eta about 1hr half. I know its setup related, so not worry about, it will be nice in future to use it on i9 with radeon vII GPU on my mac

[image]

AdrianEddy, 2022-03-01:
Fixed in b356e777e5abd7582865b0b343ef63c5da79058c

closed 2022-03-01 by AdrianEddy