gyroflow/gyroflow #246: [ubuntu] GPU encoding not possible

nightflasher, 2022-02-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0-rc4

### What operating system are you using?

ubuntu 20.04.4 LTS

### What GPU are you using?

AMD ATI Radeon R7 530 | INTEL UHD 620

### What happened?

not possible to export using GPU rendering (x264 & x265), no gpu rendering with ProRes
[image]

### Relevant log output

```shell
15:53:25 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Converting from NV12 to VAAPI_VLD
[swscaler @ 0x7f1ab366a1c0] 0bpp not supported by yuv2rgb
[swscaler @ 0x7f1ab366a1c0] vaapi_vld is not supported as output pixel format
[aac @ 0x7f1ab237a080] Qavg: nan
15:53:45 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["vdpau", "cuda", "vaapi", "qsv"]
15:53:45 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_CUDA 0
[AVHWDeviceContext @ 0x7f1ab000a080] Cannot load libcuda.so.1
[AVHWDeviceContext @ 0x7f1ab000a080] Could not dynamically load CUDA
15:53:45 [DEBUG] (2) gyroflow::rendering: Failed to create specified HW device: AV_HWDEVICE_TYPE_CUDA

15:53:45 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_VAAPI 1
15:53:45 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_VAAPI (vaapi) with format Some(AV_PIX_FMT_VAAPI_VLD)

15:53:45 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("vaapi")
15:53:45 [DEBUG] (2) gyroflow::rendering: Possible encoders with AMD: [("libx265", false)]
15:53:45 [DEBUG] (2) gyroflow::rendering: video_codec: Some("libx265"), processing_order: PreConversion
15:53:45 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 271840, render_frame_count: 6796
15:53:45 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [YUV420P, YUVJ420P, YUV422P, YUVJ422P, YUV444P, YUVJ444P, GBRP, YUV420P10LE, YUV422P10LE, YUV444P10LE, GBRP10LE, YUV420P12LE, YUV422P12LE, YUV444P12LE, GBRP12LE, GRAY8, GRAY10LE, GRAY12LE]
15:53:47 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12, BGRA, ARGB, RGBA, ABGR, BGRZ, ZRGB, RGBZ, ZBGR, X2RGB10LE, GRAY8, YUYV422, UYVY422, YUV420P, YUV420P, YUV411P, YUV422P, YUV440P, YUV444P, P010LE])
15:53:47 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Converting from NV12 to YUV420P
15:53:47 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: None, encoder_pixel_format: Some(YUV420P)
15:53:47 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P, color range: JPEG
15:53:47 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type None
```

AdrianEddy, 2022-02-25:
Please attach full log

nightflasher, 2022-02-25:
okay...just added the directory listing of lib - some files are missing
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8145165/gyroflow.log)

wkaichung, 2022-04-03:
me too! i am using Macbook Pro M1 max. Cant use GPU to export neither. same problem. The export time of a 7min clip are using 11hours to export!!!!!

AdrianEddy, 2022-04-03:
@wkaichung
Please provide log from latest dev build https://gyroflow.xyz/devbuild/?autodownload
1. Open terminal
2. Right click on Gyroflow and Open package contents -> Contents -> MacOS
3. Drag gyroflow into terminal and click enter
then after you start rendering you can cancel and close the app
gyrolow.log should be created next to gyroflow file. If it doesn't just copy me the contents of the terminal

szebedy, 2022-05-29:
Hello, I'm having the same issue with my AMD RX 550 GPU on Windows 11. The problem is that the encoder is using the CUDA library from Nvidia, which only works with Nvidia GPUs. The solution could be an alternative encoding method which does not use CUDA. I attached the error message and my log file.

[image]

[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8793218/gyroflow.log)

macrokernel, 2022-06-16:
I am having the same issue on Linux Mint 20.2 Cinnamon, kernel version: 5.13.0-41-generic https://github.com/gyroflow/gyroflow/issues/46~20.04.1-Ubuntu running on AMD Ryzen 5 4600H with Radeon Graphics (integrated graphics). 

[image]

[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8920764/gyroflow.log)

AdrianEddy, 2022-11-18:
Does this still happen in latest dev build? https://gyroflow.xyz/devbuild/?autodownload

If yes, please post the log

mituritsyn, 2022-11-18:
> Does this still happen in latest dev build? https://gyroflow.xyz/devbuild/?autodownload
> 
> If yes, please post the log

Ubuntu 20.04, AMD Ryzen 5 2600, Radeon RX580
ProRes, X264,x264 wo GPU encoding
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10039498/gyroflow.log)
[image]
after picking VAAPI

[image]

AdrianEddy, 2022-11-18:
@mituritsyn thanks for the log, I made some updates, could you please check in the latest one? https://gyroflow.xyz/devbuild/?autodownload
Please also post a log

macrokernel, 2022-11-18:
[log.txt](https://github.com/gyroflow/gyroflow/files/10043159/log.txt)

mituritsyn, 2022-11-18:
> @mituritsyn thanks for the log, I made some updates, could you please check in the latest one? https://gyroflow.xyz/devbuild/?autodownload Please also post a log

**AppImage:**
same error, but after chosing VAAPI UI blinks and I see same popup :
[image]

if I choose CPU then this appers:

[image]

moreover, sync points not showing anymore:

[image]

**Tar.Gz**
crashes after choosing CPU
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10044663/gyroflow.log)
[19518737-f690-4b40-3f1c8d8e-4835b76b.zip](https://github.com/gyroflow/gyroflow/files/10044664/19518737-f690-4b40-3f1c8d8e-4835b76b.zip)

AdrianEddy, 2022-11-19:
How about now? (same download link)

mituritsyn, 2022-11-19:
> How about now? (same download link)

seems to be fixed

mituritsyn, 2022-11-19:
> seems to be fixed

I apologize, in a hurry, almost to the end of the rendering an error appears, perhaps it is another problem, but still:
h265
[image]
h264
[image]
and then
[image]

ProRes
[image]
log attached
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10046328/gyroflow.log)

AdrianEddy, 2022-11-19:
Hmm can you try exporting just 1 second from the video? (Just set a short trim range)

ProRes doesn't support hardware acceleration except M1 macs

mituritsyn, 2022-11-19:
> Hmm can you try exporting just 1 second from the video? (Just set a short trim range)
> 
> ProRes doesn't support hardware acceleration except M1 macs

up to approximately 15 frames exporting successfully
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10047740/gyroflow.log)

nightflasher, 2022-11-19:
[gyroflow_gh886_21frames.log](https://github.com/gyroflow/gyroflow/files/10047948/gyroflow_gh886_21frames.log)
[image]
tried 21 frames, like mituritsyn with no luck

AdrianEddy, 2022-11-19:
@nightflasher your issue is a different reason, what does `vainfo` say when you type it into terminal?

AdrianEddy, 2022-11-19:
@mituritsyn How about now?

nightflasher, 2022-11-19:
`libva info: VA-API version 1.15.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.15 (libva 2.12.0)
vainfo: Driver version: Intel iHD driver for Intel(R) Gen Graphics - 22.3.1 ()
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointEncSliceLP
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointEncSliceLP
      VAProfileJPEGBaseline           :	VAEntrypointVLD
      VAProfileJPEGBaseline           :	VAEntrypointEncPicture
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointEncSliceLP
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
`
but with DRI_Prime it doesn't change, so no AMD GPU acceleration?

AdrianEddy, 2022-11-19:
Hmm so in your case, we need to tell VAAPI to use the AMD card rather than Intel, right? I think that flag is `-hwaccel_device 1` in FFmpeg CMD line?

nightflasher, 2022-11-19:
command: DRI_PRIME=1 ffmpeg -hwaccels (same without DRI_PRIME=1)

ffmpeg version 4.4.2-0ubuntu0.22.04.1 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 11 (Ubuntu 11.2.0-19ubuntu1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.22.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-librsvg --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  WARNING: library configuration mismatch
  avcodec     configuration: --prefix=/usr --extra-version=0ubuntu0.22.04.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-pocketsphinx --enable-librsvg --enable-libmfx --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared --enable-version3 --disable-doc --disable-programs --enable-libaribb24 --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libtesseract --enable-libvo_amrwbenc --enable-libsmbclient
  libavutil      56. 70.100 / 56. 70.100
  libavcodec     58.134.100 / 58.134.100
  libavformat    58. 76.100 / 58. 76.100
  libavdevice    58. 13.100 / 58. 13.100
  libavfilter     7.110.100 /  7.110.100
  libswscale      5.  9.100 /  5.  9.100
  libswresample   3.  9.100 /  3.  9.100
  libpostproc    55.  9.100 / 55.  9.100
Hardware acceleration methods:
vdpau
cuda
vaapi
qsv
drm
opencl

mituritsyn, 2022-11-20:
> @mituritsyn How about now?

Now it works. But only at half speed. Was about 40fps, now only 20fps

AdrianEddy, 2022-11-20:
@mituritsyn I believe before it wasn't encoding properly, because the HW frame pool was set to 20 frames and the frames were never freed, so it only encoded 20 frames and then it silently continued with errors until the end. You could manually time how long it takes ffmpeg to transcode the same video and how long it takes in gyroflow, should be similar. Could you confirm one last time that the encoding is working in the latest dev build? I made some changes

@nightflasher Please download latest dev build ( https://gyroflow.xyz/devbuild/?autodownload ) and let me know these:
1. output from `DRI_PRIME=0 vainfo`
2. output from `DRI_PRIME=1 vainfo`
3. GPU encoding in gyroflow using H264/AVC
4. GPU encoding in gyroflow using H265/HEVC, but adding `-hwaccel_device 1` to the "Custom encoder options"
5. Is #405 still an issue or is it fixed?

[image]

nightflasher, 2022-11-20:
1 & 2 => attached TXT
3. same as before
4. is working, but slow (about 5-10 fps)
5. i think so. can be closed (already working in stable 1.3.0 gh816)

[image]
[vainfo_both.txt](https://github.com/gyroflow/gyroflow/files/10049609/vainfo_both.txt)

AdrianEddy, 2022-11-20:
4. Could you please send me the gyroflow.log after rendering with `-hwaccel_device 1`?

nightflasher, 2022-11-20:
here you go
[gyroflow_github.log](https://github.com/gyroflow/gyroflow/files/10049990/gyroflow_github.log)

nightflasher, 2022-11-20:
But i'm trying to use the dedicated AMD GPU (specs in the first post)

AdrianEddy, 2022-11-20:
I know, please try again in latest version, i fixed one more bug. Use `-hwaccel_device 1`

macrokernel, 2022-11-20:
[image]
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10050300/gyroflow.log)

AdrianEddy, 2022-11-20:
@macrokernel does VAAPI in ffmpeg work for you? something like `ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -i INPUT_FILE.mp4 -vf "format=nv12,hwupload" -c:v h264_vaapi OUTPUT_FILE.mp4`?
What does `vainfo` return on your device? Maybe also try setting `-hwaccel_device /dev/dri/renderD128` in gyroflow encoder params.

Some more info here https://trac.ffmpeg.org/wiki/Hardware/VAAPI

nightflasher, 2022-11-20:
found something "maybe useful'
see attached file
[vainfo_libvadriver.txt](https://github.com/gyroflow/gyroflow/files/10050475/vainfo_libvadriver.txt)

mituritsyn, 2022-11-20:
> @mituritsyn Could you confirm one last time that the encoding is working in the latest dev build? I made some changes

I couldn't. Doesn't work again
[image]
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10050565/gyroflow.log)

mituritsyn, 2022-11-20:
> @macrokernel does VAAPI in ffmpeg work for you? something like `ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -i INPUT_FILE.mp4 -vf "format=nv12,hwupload" -c:v h264_vaapi OUTPUT_FILE.mp4`?

done wo errors:
`frame= 3575 fps= 23 q=-0.0 Lsize=  395219kB time=00:00:59.62 bitrate=54298.4kbits/s speed=0.384x    
video:394217kB audio:935kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.016936%
[aac @ 0x55cdac43a140] Qavg: 379.303
`
> What does `vainfo` return on your device? 

`$ vainfo
libva info: VA-API version 1.14.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/radeonsi_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: Mesa Gallium driver 22.0.5 for AMD Radeon RX 580 Series (polaris10, LLVM 13.0.1, DRM 3.42, 5.15.0-53-generic)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointEncSlice
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointEncSlice
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointEncSlice
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointEncSlice
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileJPEGBaseline           : VAEntrypointVLD
      VAProfileNone                   : VAEntrypointVideoProc
`
>Maybe also try setting `-hwaccel_device /dev/dri/renderD128` in gyroflow encoder params.
 
same error
 
[image]

AdrianEddy, 2022-11-20:
> I couldn't. Doesn't work again

Right, sorry, I made some bad changes, please try gh895 in ~10 minutes (status here https://github.com/gyroflow/gyroflow/actions/runs/3509685259)

nightflasher, 2022-11-20:
i found this "clinfo" and have the output for you
[clinfo.txt](https://github.com/gyroflow/gyroflow/files/10050740/clinfo.txt)

mituritsyn, 2022-11-20:
> > I couldn't. Doesn't work again
> 
> Right, sorry, I made some bad changes, please try gh895 in ~10 minutes (status here https://github.com/gyroflow/gyroflow/actions/runs/3509685259)

last build is ok. btw it was almost 25% faster than this one)) https://github.com/gyroflow/gyroflow/issues/246#issuecomment-1321060991

AdrianEddy, 2022-11-20:
> last build is ok. btw it was almost 25% faster than this one)) 

Cool, thanks

@nightflasher how about you in latest build? Does it work? By default or with `hwaccel_device`?

nightflasher, 2022-11-20:
no, neither off them. i have now installed ffmpeg5 (https://launchpad.net/~savoury1/+archive/ubuntu/ffmpeg5) and i have successfully rendered "ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/card0 -i test.mp4 -vf "format=nv12,hwupload" -c:v h264_vaapi test_new.mp4" with 35fps (but it's the intel graphics card). i think my dedicated amd isn't installed correctly.

i also have another terminal output with this command: **DRI_PRIME=1 LIBVA_DRIVER_NAME=radeonsi vainfo** 
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'radeonsi'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/radeonsi_drv_video.so
libva info: Found init function __vaDriverInit_1_14
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.16 (libva 2.12.0)
vainfo: Driver version: Mesa Gallium driver 23.0.0-devel for AMD Radeon R5 M445 Series (iceland, LLVM 15.0.4, DRM 3.42, 5.15.0-41-generic)
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileNone                   :	VAEntrypointVideoProc

AdrianEddy, 2022-11-20:
Did you try setting `-hwaccel_device /dev/dri/card0` in Gyroflow? Can you show me the log?

nightflasher, 2022-11-20:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10050865/gyroflow.log)
this didn't work at all.

AdrianEddy, 2022-11-20:
@nightflasher looks like your Intel card only supports H264/AVC encoding, not HEVC.
AMD may support HEVC, but doesn't look like it's detected/installed properly

Can you try again with H264/AVC?

You can select `Device for processing` under Advanced panel and select `[wgpu] AMD Radeon R5 M445 Series (RADV ICELAND) (Vulkan)`. This way Intel card will be used to encode, and AMD card will be used to process (stabilize) pixels

macrokernel, 2022-11-21:
> @macrokernel does VAAPI in ffmpeg work for you? something like `ffmpeg -hwaccel vaapi -hwaccel_device /dev/dri/renderD128 -i INPUT_FILE.mp4 -vf "format=nv12,hwupload" -c:v h264_vaapi [...]

Hi, yes, VAAPI seems to be working in ffmpeg, please see the log: 
[ffmpeg_vaapi.log](https://github.com/gyroflow/gyroflow/files/10054616/ffmpeg_vaapi.log)

Here is vainfo: 
[vainfo.txt](https://github.com/gyroflow/gyroflow/files/10054626/vainfo.txt)

Ohh, the latest Gyroflow is working with GPU acceleration! I am getting around 14.6 FPS with H264 and H265. 

[image]

AdrianEddy, 2022-11-21:
> Ohh, the latest Gyroflow is working with GPU acceleration! I am getting around 14.6 FPS with H264 and H265.

Cool, thanks for testing. You can also try selecting `[wgpu] AMD RADV RENOIR (Vulkan)`, it might be a bit faster

macrokernel, 2022-11-21:
> Cool, thanks for testing. You can also try selecting `[wgpu] AMD RADV RENOIR (Vulkan)`, it might be a bit faster

Thanks, `[wgpu] AMD RADV RENOIR (Vulkan)` is doing a bit faster - 15.4 FPS.

nightflasher, 2022-11-21:
this is getting weird: i am now back on stock ubuntu 22.04.1 LTS - no extra ffmpeg nor oibaf graphics drivers.
without the oibaf drivers, i can only select CLOVER AMD...no intel at all. 
after reading and[ comparing my gfx](https://technical.city/en/video/UHD-Graphics-620-vs-Radeon-530) options, it is weirder af: the intel UHD 620 is better than the dedicated amd r7 530 - why would a manufacturer do that? or du i miss something?
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10055262/gyroflow.log)

@macrokernel which drivers did you install?

macrokernel, 2022-11-21:
> @macrokernel which drivers did you install?

Not sure, can you please suggest a way to check the installed drivers? If I remember correct, I am using open source drivers that come with Linux Mint. 
`$ uname -a
Linux magicbook 5.15.0-52-generic #58~20.04.1-Ubuntu SMP Thu Oct 13 13:09:46 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
` 
I installed newer ffmpeg from PPA: `http://ppa.launchpad.net/savoury1/ffmpeg4/ubuntu focal main`.

AdrianEddy, 2022-11-21:
@nightflasher please try in latest version, gh899
@macrokernel  and @mituritsyn  please also test gh899 and confirm if all good

nightflasher, 2022-11-21:
Today i had time to install new drivers, so AMD drivers are up to date (again), but it's intel we wanna rock here.
So i've installed new intel gfx drivers, but gyroflow still won't use gpu to render.
[gyroflow_NV12.log](https://github.com/gyroflow/gyroflow/files/10060417/gyroflow_NV12.log)
[gyroflow_P010LE.log](https://github.com/gyroflow/gyroflow/files/10060418/gyroflow_P010LE.log)

nightflasher, 2022-11-21:
tried every option i learned the last days from you and others

nightflasher, 2022-11-21:
i had no error message and rendered was mostly with 10+ fps
but i'm stillt not sure - this could be faster?

[image]
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10060598/gyroflow.log)

AdrianEddy, 2022-11-21:
hmm that last log file seems to be truncated, did you close gyroflow before uploading the log file?

nightflasher, 2022-11-21:
no...but you will get a new one along with a screenshot
[image]
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/10060809/gyroflow.log)
 right now

AdrianEddy, 2022-11-21:
it's still truncated, you need to close gyroflow after rendering (or you can click Cancel, then close) for log to be properly finished. So far I can see that VAAPI can't be initialized at all with `-hwaccel_device 0`. Try without it

nightflasher, 2022-11-21:
i did, but the log file in the folder is the same, but i've started from comand line and there you go
[gyroflow_rendered_fine.log](https://github.com/gyroflow/gyroflow/files/10060836/gyroflow_rendered_fine.log)

nightflasher, 2022-11-21:
could this one improve the speed?
/etc/sysctl.d/99-i915.conf 
#dev.i915.perf_stream_paranoid = 0

now i've seen NV12
[gyroflow_rendered_fine_nv12.log](https://github.com/gyroflow/gyroflow/files/10060868/gyroflow_rendered_fine_nv12.log)

AdrianEddy, 2022-11-21:
`gyroflow_rendered_fine_nv12.log` looks good and it used VAAPI, how fast did it render?

nightflasher, 2022-11-21:
about 12 fps, still 80+ seconds for a 16 seconds clip - but better than 4-8fps

nightflasher, 2022-11-21:
oh and its HEVC AND H264 compatible

AdrianEddy, 2022-11-21:
well, I think there's nothing else I can do in Gyroflow, looks like it properly initializes and uses VAAPI so the rest is drivers/configuration/flags. You may still try to use the AMD card, I think it will only support H264, I recommend to first try to figure out the flags for command line `ffmpeg` and test there, then try to use these flags in Gyroflow. Good luck!

As 3 people confirmed that VAAPI now works in Gyroflow, I'm closing this issue. Thank you all for testing and providing logs, it was very helpful!

[1 more reply]

closed 2022-11-21 by AdrianEddy