gyroflow/gyroflow #249: Did anyone can stabilize footage from Insta360 One-R 1" ?

Fabersa, 2022-02-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0 rc4

### What operating system are you using?

MacOS 12.1

### What GPU are you using?

AMD FirePro D700 6gb

### What happened?

Me and some friend are trying to stabilize drone footage shooted with the Insta360 One-R 1 inch camera without success. First we pointed the problems to the drone frame, then to the camera but neither of those seems the problem who generate the wobbly video output. Anyone had success with this camera and Gyroflow?

AdrianEddy, 2022-03-02:
Yes, Insta360 One-R works fine, please post your file so we can check

Fabersa, 2022-03-11:
Can anyone try to stabilize this sample video from my One-R to check if I'm doing wrong or somethings fails on my camera? It's 5.7k ultrawide. Thanks
https://drive.google.com/file/d/1_JNonLLmzi0erD9bBVh5E5HQz2supLCL/view?usp=sharing

dronotron, 2022-03-16:
just calibrate your camera gyro before your flight

AdrianEddy, 2022-04-13:
The file itself works fine in Gyroflow, but you have a lot of noise in the gyro data - it's a hardware issue, maybe you need to soft mount the camera or fine tune your frame. 
I also tried in Insta360 studio and it also can't handle these vibrations well, so there's nothing more Gyroflow can do here.