gyroflow/gyroflow #250: Add Preview and Export LUTs Function

HerbertEttore, 2022-02-27:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Most of us use "FLAT" or "LOG" color profile on our cameras to get the best results in post production.

But sometimes we want to export a quick video of that amazing flight or scene to post on Instagram, Youtube or any other platform. But exporting in Gyroflow and then putting in the editing app and then exporting again can be a bit time consuming.

So it would be very convenient if Gyroflow had a function to apply a ready-to-use LUT. Thus halving the time for a quick video.

I did some testing with FFmpeg as following command and it worked very well:
`-vf lut3d='D\:/Insta360-LUT/ONE-R-LUT/4K-WIDE-ANGLE.CUBE'`
[SMO 4k (ONE R) Original LUT](https://res.insta360.com/static/assets/storage/20210625/7936eff98ff1e6eac434c009824713ee/Insta360-LUT.zip)

Would this implementation be possible?

ghuser123321, 2022-04-07:
I'd honestly prefer to see the OFX version catch up, rather than putting video editor things in the stand alone.

rgaufman, 2025-09-03:
I created a simple script that with a single command stabilises and compresses all my videos, I use this gyroflow template:

```
{
  "version": 2,
  "output": {
    "audio": true,
    "audio_codec": "AAC",
    "codec": "H.265/HEVC",
    "encoder_options": "-crf 24 -pix_fmt yuv422p10le -profile:v main422-10 -x265-params log-level=error:range=full -vf lut3d='/Users/hackeron/sony_lut_33.cube'",
    "export_trims_separately": false,
    "interpolation": "lanczos4",
    "output_filename": "",
    "pad_with_black": false,
    "preserve_other_tracks": false,
    "use_gpu": false
  },
  "stabilization": {
    "method": "Default",
    "fov": 1.0,
    "adaptive_zoom_method": 1,
    "adaptive_zoom_window": 4,
    "max_zoom": 120,
    "max_zoom_iterations": 5,
    "horizon_lock_amount": 0,
    "horizon_lock_roll": 0,
    "lens_correction_amount": 1,
    "smoothing_params": [
      {
        "name": "smoothness",
        "value": 0.5
      },
      {
        "name": "smoothness_pitch",
        "value": 0.5
      },
      {
        "name": "smoothness_yaw",
        "value": 0.5
      },
      {
        "name": "smoothness_roll",
        "value": 0.5
      },
      {
        "name": "per_axis",
        "value": 0
      },
      {
        "name": "trim_range_only",
        "value": 1
      },
      {
        "name": "max_smoothness",
        "value": 1
      },
      {
        "name": "alpha_0_1s",
        "value": 0.1
      }
    ],
    "frame_readout_direction": "TopToBottom",
    "frame_readout_time": 9.56574171757578
  },
  "trim_ranges_ms": [
    [
      500,
      758259
    ]
  ]
}
```

This works great, but the LUT is ignored as that's an ffmpeg parameter, not an x265 one. Is there any way to allow passing a LUT file? 

If not, I suppose the workaround is to use a low compression export like ProRes and then do a second pass with ffmpeg, but that introduces a lot of additional processing time.

AdrianEddy, 2025-09-03:
it's not possible to add any ffmpeg filters this way, these parameters are for encoder only
and it's not easy to add ffmpeg filters in the pipeline because filters need RGB and we're processing YUV planes separately

I'll implement proper support for LUTs some day

requies, 2026-01-12:
> it's not possible to add any ffmpeg filters this way, these parameters are for encoder only and it's not easy to add ffmpeg filters in the pipeline because filters need RGB and we're processing YUV planes separately
> 
> I'll implement proper support for LUTs some day

 Hi, is there any news?

rsmith4321, 2026-10-07:
I added this feature to my Gyroflow fork for my own workflow, and it’s been working well with my DJI O4 Pro footage.
Under Export settings, there’s a .cube LUT selector and simple brightness and contrast sliders. The LUT and adjustments are shown in the live preview and applied to the final stabilized export. You can also toggle the preview colors for comparison and double-click either slider to reset it.
I’ve mainly tested with DJI’s O4 D-Log M to Rec.709 LUT, so I can’t speak for every LUT or camera profile yet. I developed it with Codex assistance and have been reviewing and testing the changes.
I use it alongside my [Easy Eject app](https://github.com/rsmith4321/Ejector), which automatically imports and organizes my drone footage and makes it easy to open in Gyroflow. Since I usually shoot in log, this gives me a quicker workflow: import, stabilize, select the LUT, and export—without needing another editing app for clips that only need basic color correction.
The changes are available in [PR #1249](https://github.com/gyroflow/gyroflow/pull/1249). They’re in my fork for now and haven’t been merged into the official Gyroflow release.
Feedback would be welcome, especially from anyone trying other LUTs or camera profiles!

rsmith4321, 2026-10-10:
Following up on my earlier LUT work: the project has grown into **Gyroflow Plus**, an open-source community fork of Gyroflow focused on stabilizing and finishing log footage in one app. Version 1.0 for Mac is now available.

Alongside Gyroflow's stabilization and trimming workflow, it adds LUT selection, recent LUTs, and exposure, temperature, tint, brightness, contrast, highlights, shadows, and saturation controls. Colors appear in the preview and are applied to the stabilized export. Originals remain untouched, and saved projects keep the settings editable.

The color pipeline uses **OpenColorIO**, with its CPU processor for export and generated GPU processing for the preview. Our own code connects it to Gyroflow's video pipeline and controls. This also performs better than the early implementation.

I chose to develop a fork because the work grew beyond the original LUT and brightness/contrast patch into a broader color workflow, with its own testing and packaging needs. Keeping it separate lets me iterate and ship builds independently. It builds on Gyroflow's work and isn't an official Gyroflow release; focused upstream contributions are still welcome.

**Download (Mac with Apple silicon, signed and notarized):** https://github.com/rsmith4321/gyroflow-plus/releases/tag/plus-v1.0.0. The full source is attached to the release.
**App information:** https://gyroflowplus.com/
**Source:** https://github.com/rsmith4321/gyroflow-plus

I've mainly tested with DJI O4 Pro D-Log M footage. The fork keeps Gyroflow's camera support, but that doesn't mean every camera or color profile has been tested with LUTs. Use a LUT made for your footage. This is non-destructive video grading, not RAW recovery. Windows builds aren't available yet.

The earlier [PR #1249](https://github.com/gyroflow/gyroflow/pull/1249) was closed because this larger effort superseded its limited implementation. Its code and discussion remain available. Includes AI-assisted code, reviewed and tested.

Feedback and reproducible reports are welcome at https://github.com/rsmith4321/gyroflow-plus/issues, especially for other cameras and LUTs.

rsmith4321, 2026-10-10:
Quick update: Gyroflow Plus is now called **GyroGrade**, and version 1.0.1 for Mac is out.

Why a fork instead of waiting for this to land here: we wanted LUT support for log footage right away, and official Gyroflow releases come out fairly infrequently, so we decided not to wait. A fork let us ship it now and keep adding improvements. Alongside the LUT, there are exposure, temperature, tint, brightness, contrast, highlights, shadows and saturation controls, all shown in the preview and applied to the stabilized export.

**Download (Mac with Apple silicon, signed and notarized):** https://github.com/rsmith4321/gyrograde/releases/tag/plus-v1.0.1
**Website:** https://gyrograde.com/
**Source:** https://github.com/rsmith4321/gyrograde

If you installed Gyroflow Plus 1.0, GyroGrade copies your settings and recent LUTs on first launch. It's based on Gyroflow but isn't an official Gyroflow release, so please report problems at https://github.com/rsmith4321/gyrograde/issues rather than here. Thanks to everyone in this thread for asking for LUTs, and to the Gyroflow team for the work this builds on.

HerbertHTSE, 2026-10-16:
> Quick update: Gyroflow Plus is now called **GyroGrade**, and version 1.0.1 for Mac is out.
>
> Why a fork instead of waiting for this to land here: we wanted LUT support for log footage right away, and [...]

That's what I was talking about! 

I tested your fork here and it worked very well.
That's exactly what I was thinking initially.

Congratulations on the great work!

requies, 2026-10-17:
> 15\. Quick update: Gyroflow Plus is now called **GyroGrade**, and version 1.0.1 for Mac is out.
>      Why a fork instead of waiting for this to land here: we wanted LUT support for log footage right away, and official Gyroflow releases come out fairly infrequently, so we decided not to wait. A fork let us ship it now and keep adding improvements. Alongside the LUT, there are exposure, temperature, tint, brightness, contrast, highlights, shadows and saturation controls, all shown in the preview and applied to the stabilized export.
>      **Download (Mac with Apple silicon, signed and notarized):** https://github.com/rsmith4321/gyrograde/releases/tag/plus-v1.0.1
>      **Website:** https://gyrograde.com/
>      **Source:** https://github.com/rsmith4321/gyrograde
>      If you installed Gyroflow Plus 1.0, GyroGrade copies your settings and recent LUTs on first launch. It's based on Gyroflow but isn't an official Gyroflow release, so please report problems at https://github.com/rsmith4321/gyrograde/issues rather than here. Thanks to everyone in this thread for asking for LUTs, and to the Gyroflow team for the work this builds on.

I had lost track of this post. I built it on my Linux (CachyOS) machine and it works great. 
Love the LUT/Colors settings panel.
Thank you.

rsmith4321, 2026-10-17:
Thanks, please report any issues and let me know. Of course, if anyone wants to port it back into the official project please do. But I think a port was the fastest way to get it out and I had fun doing it. Of course I had help from Claude, but who doesn't now? Open source is great.