gyroflow/gyroflow #253: Add command line interface

adworacz, 2022-03-04:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Gyroflow is incredible, first and foremost. 

I find myself needing to process files in batch though, specifically to remove fish eye distortion. Gyroflow has the best distortion removal I’ve been able to find. It would be great if it supported a CLI so I could process a bunch of files in one fell swoop. 

Something like:
```
gyroflow -i input.mp4 —remove-fisheye out.mp4
```

Obviously, other options will be helpful, like pointing to a lens profile, or using auto sync for stabilization etc. 

Just wanted to post up this feature request, as I’m sure there are others who’d love the ability to automate much of their workflow.

ElvinC, 2022-03-04:
Duplicate of https://github.com/gyroflow/gyroflow/issues/27

Thanks for the suggestion though!

closed 2022-03-04 by ElvinC

adworacz, 2022-03-04:
Ah crap don’t know how I missed that, I even searched for it too. 

Sorry for the duplicate!

AdrianEddy, 2022-09-01:
Done in 3c4219d2eb4b4f8c3ee02b72ba9b3abc2dc08dc2 and 1ff6506219683b35e90f1a076b80895eccbe5289
Available in latest dev build: https://gyroflow.xyz/devbuild/?autodownload
```
Usage: gyroflow.exe [<input...>] [-f] [-j <parallel-renders>] [-d <when-done>] [-p <out-params>]

Positional Arguments:
  input             input files: videos, project files, lens profiles, presets

Options:
  -f, --overwrite   overwrite if output file exists, default: false
  -j, --parallel-renders
                    number of parallel renders, default: 1
  -d, --when-done   when done: 1 - shut down; 2 - reboot; 3 - sleep; 4 -
                    hibernate; 5 - logout
  -p, --out-params  output parameters, eg. "{ 'codec': 'H.265/HEVC', 'bitrate':
                    150, 'use_gpu': true, 'audio': true }"
  --help            display usage information
```

Basically provide video or project file, and optionally you can specify lens profile file and/or preset file. It supports rendering multiple files in a single call.

If any functionality is missing, let me know your use case.