gyroflow/gyroflow #255: Gyroflow crashes when importing video

aerevides, 2022-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0

### What operating system are you using?

Windows 10 Pro

### What GPU are you using?

Nvidia Geforce GTX 1050

### What happened?

The app crashes when importing files bigger than 4 gb

### Relevant log output

```shell
20:40:43 [DEBUG] (6) gyroflow::util: Minidump uploaded: OK
20:40:44 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\Utente\\Desktop\\Gyroflow-windows64\\camera_presets"
20:40:44 [INFO] Loaded 1293 lens profiles in 22.951ms
20:40:44 [INFO] Latest version: v1.0.0-rc4, current version: 1.0.0-rc4
20:40:44 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 1050
20:40:44 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
20:40:44 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src\rendering\mod.rs:47] gpu_type = NVIDIA
20:40:54 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
20:40:54 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.video" ""
20:40:54 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
20:40:54 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "thread.subtitle" ""
20:40:55 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "decoder.subtitle" "FFmpeg"
20:40:55 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
20:40:55 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
20:40:55 [DEBUG] (14) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
20:40:55 [DEBUG] (14) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x11510b1a0b0) QSize(1289, 725)
20:40:55 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.D3D11"
20:40:55 [DEBUG] (14) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
20:40:55 [INFO] surface size: 1280x720
Crashdump written to C:\Users\Utente\Desktop\Gyroflow-windows64/a4a4b538-d734-413c-a3c6-137aaacb7bed.dmp
Premere un tasto per continuare . . .
```

AdrianEddy, 2022-03-06:
Please upload a file which causes the crash

AdrianEddy, 2022-04-13:
@aerevides Please upload a file which causes the crash

AdrianEddy, 2022-04-28:
Please reopen if this still happens in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-04-28 by AdrianEddy