gyroflow/gyroflow #256: Black pixels in white areas of the output video

avsaase, 2022-03-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0-RC4

### What operating system are you using?

Windows 10

### What GPU are you using?

Radeon RX 570

### What happened?

My output video has some black pixels in white areas of the frame (that are not in the input video):
[image]
[image]

This happens both with CPU and GPU encoding but the black pixels are not visible in the video preview. 

Here are the original and processed videos: https://drive.google.com/drive/folders/19W-V0fH5X4CkeF3TJElUFNdRXscMtdXq?usp=sharing. The problem is best visible from 1:15 onwards.

The camera is a Runcam 5 Orange. Please let me know if you need more info.

AdrianEddy, 2022-03-07:
Can you reproduce it using latest dev build? Can you attach gyroflow.log after exporting with these artefacts? I've seen this before but I can't seem to reproduce it anymore, weird

avsaase, 2022-03-07:
Yes, same issue with the latest dev build. I also tried h264 encoding but that didn't make a difference. 

I added the gyroflow.log file to the Google Drive folder.

AdrianEddy, 2022-03-07:
Fixed in c360a63d1275db0dce33e2dbdc1a96660a8e7de6

closed 2022-03-07 by AdrianEddy