gyroflow/gyroflow #258: No GPU acceleration on Mac OS?

linxucc, 2022-03-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0 rc4

### What operating system are you using?

macOS BigSur 11.6.4

### What GPU are you using?

AMD Radeon Pro 570

### What happened?

Hi, after fiddling with the settings within the App I found that when exporting there's no GPU involved in Mac OS?

I tried to stabilize my video from GoPro Hero 10, at 5.3K@60fps, it tells me a 30 seconds clip would take 2 hrs to export... 

And it seems there's no GPU involvement in the process.....

[image]

I searched and noticed that there are already other issues about this, but have to add my 2 cents because I think it's important.

As you can see, without hardware acceleration this software is not usable, unless you are running in in a super computer or only process simple 1080p or below in h264.

Besides, before I turned to github, I read the website thoroughly but found nothing about this, it says "all platform supports GPU acceleration", which is obviously misleading by anyway...

I think if this problem cannot be fixed in short time, the website should change to reflect this. It could save people a lot of time trying to figure out this. Many could just turn to a windows PC saving a lot of time troubleshooting.

AdrianEddy, 2022-03-07:
It's already fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-03-07 by AdrianEddy

linxucc, 2022-03-07:
Hi! Thanks for your effort!

But I have just tried the latest build, some part of it has been fixed but some are not.

What now is OK:
1. Preview, smoother than before (I don't know if it's placebo), GPU usage at some degree.
2. Export to ProRes is OK, no matter the quality. GPU usage very good, speed is good.

What now is not OK:
1. Export to x265. There's no difference than the non-dev version. 

You could take another look at it? Maybe x265 got missed or something?

linxucc, 2022-03-07:
This is utilization when exporting to ProRes, in case you want to know. It's really good. 
But x265 still not working.

[image]

AdrianEddy, 2022-03-07:
Can you test also x264? It's possible your GPU doesn't have hardware x265
encoder

linxucc, 2022-03-07:
This is the x265 export, obviously no GPU involved, takes forever.
[image]

This is x264, GPU is involved.
[image]

My computer is iMac 5k 2017 base model, with AMD Radeon Pro 570. As of whether there is GPU accelerated x265 encoding, It's a little hard to find out, since it's not very clear mentioned both Apple and AMD. The decoding part is obviously OK, but encoding part is mysterious. And the videoToolbox on MacOS definitely makes things a little bit more complicated.

I did various tests though, using the x264 file Gyroflow just generated, transcode it into h265:
1. Using Apple's "Compressor" (hevc 8bit): seems some degree of hardware acceleration, completed in about 13 mins.
[image]
2. Using Adobe's "Media Encoder": seems some degree of hardware acceleration, also completed in about 13 mins.
[image]
3. Handbrake with x265, it's software, CPU flatlined, takes forever.
4. Handbrake with x265(videoToolbox), same as 1 and 2, some degree of acceleration, 13 minutes around.
5. Apple's Compressor (hevc 10bit): no hardware acceleration, takes forever.

So on the Mac side things are a little bit complicated though, seems on my computer there is hardware accelerated hevc 8bit encoding, but since Apple's videoToolbox is a blackbox, we don't know for sure how it's done, it could be the AMD GPU or the Intel integrated GPU, or something else.
But I think if videoToolbox is properly used, this problem could be fixed.

AdrianEddy, 2022-03-07:
Thanks for detailed analysis, Gyroflow uses VideoToolbox for x264, x265 and ProRes, but it's possible it fails to initialize or something. Can you run gyroflow in terminal and copy the log here after exporting? (Open package contents and run Contents/MacOs/gyroflow in terminal)

linxucc, 2022-03-07:
I found the log and it reads like this:

11:23:06 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("videotoolbox")
11:23:06 [DEBUG] (2) gyroflow::rendering: Possible encoders with AMD: [("hevc_videotoolbox", true), ("libx265", false)]
11:23:06 [DEBUG] (2) gyroflow::rendering: video_codec: Some("hevc_videotoolbox"), processing_order: PreConversion
11:23:06 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 246963, render_frame_count: 14803
11:23:06 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [VIDEOTOOLBOX, NV12, YUV420P, BGRA, P010LE]
11:23:06 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12])
11:23:06 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_VIDEOTOOLBOX), encoder_pixel_format: Some(NV12)
11:23:06 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: NV12, color range: JPEG
11:23:06 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_VIDEOTOOLBOX)
11:23:06 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x7fbc7660a540
11:29:00 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["videotoolbox"]
11:29:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_VIDEOTOOLBOX 0
11:29:00 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_VIDEOTOOLBOX (videotoolbox) with format Some(AV_PIX_FMT_VIDEOTOOLBOX)

I'll try run directly from terminal now.

linxucc, 2022-03-07:
Directly run from terminal, this is what shows in the terminal:

DK 0.14.0 (git da2fab7) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS1090 x86_64; AppleClang13.0.0 (clang-1300.0.29.30)GCC4.2.1; libc++12000; 15:15:44 Mar  6 2022
12:53:40 [WARN] Populating font family aliases took 558 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
12:53:43 [INFO] Lens profiles directory: "/Applications/Gyroflow.app/Contents/Resources/camera_presets"
12:53:43 [INFO] Latest version: v1.0.0-rc4, current version: 1.0.0-rc4 (gh359)
12:53:44 [INFO] Loaded 1293 lens profiles in 74.637ms
12:53:44 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Apple") ["Intel(R) Core(TM) i5-7500 CPU @ 3.40GHz", "AMD Radeon Pro 570 Compute Engine"]
12:53:44 [INFO] OpenCL Platform: Apple, Device: AMD AMD Radeon Pro 570 Compute Engine
12:53:44 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_D3D11VA
12:53:44 [DEBUG] (1) gyroflow::rendering: Failed to create specified HW device: AV_HWDEVICE_TYPE_D3D11VA

[src/rendering/mod.rs:47] gpu_type = AMD
12:54:42 [DEBUG] (6) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
12:54:42 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "thread.video" ""
12:54:42 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
12:54:42 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
12:54:42 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
12:54:42 [DEBUG] (7) MDKPlayer: [in m_player->onEvent] "decoder.video" "VideoToolbox"
AudioQueuePause @4342.000 (sample time)/22485454993958 (host time)
12:54:42 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Metal
12:54:42 [DEBUG] (9) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7f9a53c8cf80) QSize(1546, 870)
12:54:43 [DEBUG] (9) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
12:54:43 [INFO] surface size: 1920x1080
AudioQueuePause @11141.000 (sample time)/22486279823796 (host time)
AudioQueuePause @18448.000 (sample time)/22486453467011 (host time)
12:54:43 [DEBUG] (9) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Metal
12:54:43 [DEBUG] (9) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7f9a54898c90) QSize(1920, 1080)
AudioQueuePause @25781.000 (sample time)/22487405521106 (host time)
2022-03-07 20:54:48.015 gyroflow[3473:238797] TSM AdjustCapsLockLEDForKeyTransitionHandling - _ISSetPhysicalKeyboardCapsLockLED Inhibit
12:55:57 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["videotoolbox"]
12:55:57 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_VIDEOTOOLBOX 0
12:55:57 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_VIDEOTOOLBOX (videotoolbox) with format Some(AV_PIX_FMT_VIDEOTOOLBOX)

12:55:57 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("videotoolbox")
12:55:57 [DEBUG] (2) gyroflow::rendering: Possible encoders with AMD: [("hevc_videotoolbox", true), ("libx265", false)]
[src/rendering/ffmpeg_hw.rs:174] &dev.hw_formats = [
    VIDEOTOOLBOX,
]
[src/rendering/ffmpeg_hw.rs:175] &dev.sw_formats = [
    NV12,
    AYUV64LE,
    YUV420P,
    UYVY422,
    P010LE,
    NV16,
    P210LE,
    P216LE,
    NV24,
    P410LE,
    P416LE,
    BGRA,
]
12:55:57 [DEBUG] (2) gyroflow::rendering: video_codec: Some("hevc_videotoolbox"), processing_order: PreConversion
12:55:57 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 246963, render_frame_count: 14803
12:55:57 [DEBUG] (2) gyroflow::rendering::ffmpeg_processor: Codec formats: [VIDEOTOOLBOX, NV12, YUV420P, BGRA, P010LE]
12:55:58 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: Some([NV12])
12:55:58 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_VIDEOTOOLBOX), encoder_pixel_format: Some(NV12)
12:55:58 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: NV12, color range: JPEG
12:55:58 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type Some(AV_HWDEVICE_TYPE_VIDEOTOOLBOX)
12:55:58 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: Setting hw_device_ctx 0x7f9a5480b840
[aac @ 0x7f9a54d26340] Qavg: 225.869
AudioQueuePause @374092.000 (sample time)/22729677245508 (host time)

AdrianEddy, 2022-03-07:
hmm, the log you provided indicates the file is 8-bit (format NV12) and that it uses VideoToolbox, so I'm not sure what went wrong but I'm afraid it's out of our control - Gyroflow just sends the frames to VideoToolbox and it handles the rest

linxucc, 2022-03-10:
I read the code, it seems Gyroflow indeed just send whatever it has to ffmpeg than VideoToolbox, Gyroflow did nothing strange. I'm pretty sure it's a 8-bit 10-bit problem, and it happens either at ffmpeg's or videotoolbox's "guess work", people seems to have all kinds of issue just like this.

I think specifically set the parameters to ffmpeg could solve this, but that's too far a reach for now. I myself too just give up and use other format instead. I think if in the future there are people encounter same issue, you can relate to this.

Thanks for your effort in this! This software is brilliant!