gyroflow/gyroflow #261: Folder access request keep popping up

L1cardo, 2022-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0-RC5

### What operating system are you using?

macOS 12.2.1 (21D62)

### What GPU are you using?

Mac mini M1

### What happened?

Folder access request keep popping up even after I clicked allow button.
[image]

When I tried to open Gyroflow via terminal, this problem won't happen!!

[image]

### Relevant log output

```shell
Last login: Tue Mar  8 01:04:03 on ttys001
licardo@LicardodeMac-mini Desktop % gyroflow
zsh: command not found: gyroflow
licardo@LicardodeMac-mini Desktop % /Applications/Gyroflow.app 
zsh: permission denied: /Applications/Gyroflow.app
licardo@LicardodeMac-mini Desktop % /Applications/Gyroflow.app/Contents/MacOS/gyroflow
MDK 0.14.0 (git da2fab7) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS110000 ARMv8 AArch64; AppleClang13.0.0 (clang-1300.0.29.30)GCC4.2.1; libc++12000; 15:17:19 Mar  6 2022
03:16:02 [WARN] Populating font family aliases took 109 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
03:16:03 [INFO] Lens profiles directory: "/Applications/Gyroflow.app/Contents/Resources/camera_presets"
03:16:03 [INFO] Loaded 1293 lens profiles in 13.998ms
03:16:03 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Apple") ["Apple M1"]
03:16:03 [INFO] OpenCL Platform: Apple, Device: Apple Apple M1
03:16:03 [WARN] Unknown GPU apple apple m1
[src/rendering/mod.rs:47] gpu_type = Unknown
03:16:04 [INFO] Latest version: v1.0.0-rc5, current version: 1.0.0-rc5
2022-03-09 11:16:33.853 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:33.853 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:33.854 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:33.854 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:33.855 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:33.855 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.531 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.531 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.532 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.532 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.537 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.537 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
licardo@LicardodeMac-mini Desktop % clear

licardo@LicardodeMac-mini Desktop % /Applications/Gyroflow.app/Contents/MacOS/gyroflow
MDK 0.14.0 (git da2fab7) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS110000 ARMv8 AArch64; AppleClang13.0.0 (clang-1300.0.29.30)GCC4.2.1; libc++12000; 15:17:19 Mar  6 2022
03:16:45 [WARN] Populating font family aliases took 110 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
03:16:46 [INFO] Lens profiles directory: "/Applications/Gyroflow.app/Contents/Resources/camera_presets"
03:16:46 [INFO] Loaded 1293 lens profiles in 18.786ms
03:16:46 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Apple") ["Apple M1"]
03:16:46 [INFO] OpenCL Platform: Apple, Device: Apple Apple M1
[... 23 lines cut ...]
2022-03-09 11:16:33.854 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:33.854 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:33.855 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:33.855 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.531 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.531 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.532 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.532 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.537 gyroflow[32434:11374207] -[NSNull length]: unrecognized selector sent to instance 0x207684340
2022-03-09 11:16:36.537 gyroflow[32434:11374207] Exception calling directoryDidChange: on delegate <QNSOpenSavePanelDelegate: 0x600002d848a0> exception=-[NSNull length]: unrecognized selector sent to instance 0x207684340
licardo@LicardodeMac-mini Desktop % clear

licardo@LicardodeMac-mini Desktop % /Applications/Gyroflow.app/Contents/MacOS/gyroflow
MDK 0.14.0 (git da2fab7) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS110000 ARMv8 AArch64; AppleClang13.0.0 (clang-1300.0.29.30)GCC4.2.1; libc++12000; 15:17:19 Mar  6 2022
03:16:45 [WARN] Populating font family aliases took 110 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
03:16:46 [INFO] Lens profiles directory: "/Applications/Gyroflow.app/Contents/Resources/camera_presets"
03:16:46 [INFO] Loaded 1293 lens profiles in 18.786ms
03:16:46 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Apple") ["Apple M1"]
03:16:46 [INFO] OpenCL Platform: Apple, Device: Apple Apple M1
03:16:46 [WARN] Unknown GPU apple apple m1
[src/rendering/mod.rs:47] gpu_type = Unknown
03:16:47 [INFO] Latest version: v1.0.0-rc5, current version: 1.0.0-rc5
licardo@LicardodeMac-mini Desktop %
```

AdrianEddy, 2022-04-13:
Can you please try in latest dev build? https://gyroflow.xyz/devbuild/?autodownload

L1cardo, 2022-04-19:
> Can you please try in latest dev build? [gyroflow.xyz/devbuild/?autodownload](https://gyroflow.xyz/devbuild/?autodownload)

This build does not have this problem. Thanks for your work!

closed 2022-04-19 by AdrianEddy