gyroflow/gyroflow #263: Radeon Pro Vega 64 16 GB GPU encoding crashes

lomk07, 2022-03-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0

### What operating system are you using?

mac os 10.14.6

### What GPU are you using?

Radeon Pro Vega 64 16 GB

### What happened?

GPU / Hardware acceleration works in preview but not with rendering. ffmpeg h264/h265 and prores enabled.
Happens with both GoPro 9 and IFlight/Insta360 Go P3 at 4K. Exporting without GPU encoding works.

Nevertheless THANK you devs for providing this amazing tool - FOR FREE. You rock!

[image]

AdrianEddy, 2022-03-10:
Did you try disabling GPU decoding under Advanced?

lomk07, 2022-03-10:
I´m using 1.0.0 rc5 on mac. There are two options: GPU encoding for exporting / GPU decoding for preview window (as far as I understand the UI - please correct me if I´m wrong) 

When I try to export with encoding on and decoding off a slightly different error is displayed

[image]
.

I´ll try to get a better log later.

AdrianEddy, 2022-03-10:
What are your export settings? Which codec and bitrate? What format is the input file?

AdrianEddy, 2022-04-13:
Please attach `gyroflow.log` after you get that error and close the app

Right click on Gyroflow -> Open package contents -> Contents -> MacOS -> gyroflow.log

AdrianEddy, 2022-04-30:
Should be fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-04-30 by AdrianEddy