gyroflow/gyroflow #268: RC5 crashing on open file with Insta360 One R 1inch files

rodrigosclosa, 2022-03-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0-RC5

### What operating system are you using?

Windows 10 Pro

### What GPU are you using?

NVIDIA GeForce 1060 6GB

### What happened?

When I try to open the files fomr One R 1Inch, it crashes... I've tried with many files, same issue..

Video files:
https://drive.google.com/drive/folders/1GZ8KST1QUnIkaaFNEz2b-66kc4N_tf5j?usp=sharing

### Relevant log output

```shell
15:36:38 [INFO] [MDK] MDK 0.14.0 (git da2fab7) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Windows0x0600 x86_64; MSVC1931; MSVCRT14.31.31103.0/msvcp140.dll; 15:20:47 Mar  6 2022
15:36:38 [DEBUG] (1) mdk: [MDK] 3mdk verify key signature ok
15:36:38 [DEBUG] (1) mdk: [MDK] 3mdk license key restrictions: OpenSource,
15:36:38 [DEBUG] (1) mdk: [MDK] user module: D:\Downloads\Gyroflow-windows64\Gyroflow.exe
15:36:38 [DEBUG] (1) mdk: [MDK] user module: D:\Downloads\Gyroflow-windows64\Gyroflow.exe
15:36:38 [DEBUG] (1) mdk: [MDK] 3mdk license key for app: gyroflow
15:36:41 [INFO] Lens profiles directory: "\\\\?\\D:\\Downloads\\Gyroflow-windows64\\camera_presets"
15:36:41 [INFO] Latest version: v1.0.0-rc5, current version: 1.0.0-rc5
15:36:41 [INFO] Loaded 1293 lens profiles in 81.675ms
15:36:41 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") ["NVIDIA GeForce GTX 1060"]
15:36:41 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") ["Intel(R) HD Graphics 630"]
15:36:41 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 1060
15:36:41 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
15:36:41 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
15:36:45 [DEBUG] (1) mdk: [MDK] Registered audio backends: XAudio2 DSound
15:36:45 [DEBUG] (1) mdk: [MDK] XAudio2_9.dll
15:36:45 [DEBUG] (1) mdk: [MDK] Build with XAudio2 from Win8+ SDK
15:36:45 [DEBUG] (1) mdk: [MDK] Try symbol 'XAudio2Create' from Windows SDK dll
15:36:45 [DEBUG] (1) mdk: [MDK] Registered audio backends: XAudio2 DSound
15:36:45 [DEBUG] (1) mdk: [MDK] 00000258DEF124F0__cdecl mdk::abi::Player::Player(void)
15:36:45 [INFO] [MDK] 00000258D6EB8F40 MediaControl.prepare(0, ...) I:/DCIM/Camera01/PRO_VID_20211217_194636_00_006.mp4
15:36:45 [DEBUG] (1) mdk: [MDK] 0000025884764AC0 FrameReaderImpl.start(0, ...)
15:36:45 [DEBUG] (10) mdk: [MDK] start frame reader thread: 23648
15:36:45 [DEBUG] (10) mdk: [MDK] try to load av module: D:\Downloads\Gyroflow-windows64\ffmpeg-5.dll
15:36:45 [DEBUG] (10) mdk: [MDK] try to load av module: D:\Downloads\Gyroflow-windows64\avdevice-59.dll
15:36:45 [DEBUG] (10) mdk: [MDK] Selected avdevice runtime version: 59.4.100 (build: 59.5.100), license: GPL version 2 or later
15:36:45 [DEBUG] (10) mdk: [MDK] Selected avdevice runtime configuration: --extra-version=QtAV --disable-doc --disable-debug --enable-shared --enable-runtime-cpudetect --enable-libmfx --disable-ptx-compression --enable-mediafoundation --disable-postproc --ranlib=llvm-ranlib-13 --nm=llvm-nm-13 --ar=llvm-ar-13 --cc=clang-13 --ld='/usr/lib/llvm-13/bin/lld-link ' --nvcc=clang-13 --enable-cross-compile --arch=x86_64 --target-os=win32 --disable-stripping --extra-cflags='--target=x8
15:36:45 [DEBUG] (10) mdk: [MDK] 6_64-pc-windows-msvc -DWIN32 -D_WIN32 -D_WIN32_WINNT=0x0600 -Wno-nonportable-include-path -Wno-deprecated-declarations -I/tmp/dep/include -Xclang -ivfsoverlay -Xclang "/tmp/winsdk/vfs.yaml" -g -gcodeview -Xclang -cfguard' --extra-ldflags='-DEBUG -OPT:REF -SUBSYSTEM:CONSOLE -NODEFAULTLIB:libcmt -DEFAULTLIB:msvcrt -guard:cf' --extra-libs=oldnames.lib --enable-small --disable-filters --enable-filter='*null*,*fade,*fifo,*format,*resample,aeval,all
15:36:45 [DEBUG] (10) mdk: [MDK] *,atempo,pan,color*,convolution,crop,draw*,eq*,framerate,*_cuda,*_qsv,*_vaapi,*v4l2*,hw*,scale,volume' --disable-muxers --disable-encoders --disable-decoders --enable-decoder='*sub*,movtext,*web*,aac*,*ac3*,alac*,ape,ass,av1*,cc_dec,cook,dca,dnxhd,eac3*,exr,truehd,ff*,*yuv*,flv,flac,gif,h26[3-4]*,hevc*,hap,mp[1-3]*,prores,*peg*,mlp,mpl2,nellymoser,opus,pcm*,*png*,rawvideo,rv*,sami,srt,ssa,v210*,vc1*,vorbis,vp[6-9]*,wm*,wrapped_avframe' --disab
15:36:45 [DEBUG] (10) mdk: [MDK] le-demuxers --enable-demuxer='*sub*,*ac3,*ac,*peg*,*web*,ape,ass,avi,concat,dnxhd,dts*,*dash*,*flv,gif,hls,h264,hevc,kux,xv,matroska,mlv,mov,mp3,mxf,nsv,nut,ogg,pcm*,rawvideo,rt*p,spdif,srt,vc1,v210*,wav,*pipe,image2' --disable-parsers --enable-parser='*sub*,aac*,ac3,cook,dnxhd,flac,h26[3-4],hevc,m*,opus,rv*,vc1,vorbis,vp[8-9]' --pkg-config=pkg-config --pkg-config-flags=--static --enable-gpl --enable-libx265 --enable-libx264 --enable-encoder='
15:36:45 [DEBUG] (10) mdk: [MDK] libx26*,aac,dnxhd,exr,ff*,*yuv*,gif,h26[3-4]*,hevc*,mjpeg,*png,opus,pcm*,prores*,rawvideo,spdif,speedhq,*jpeg,*png,vp[7-9],wrapped_avframe,nvenc*,vorbis' --enable-muxer='libx26*,*jpeg,dnxhd,fifo,flv,gif,hls,h264,hevc,image2,mov,mp4,mpegts,matroska,og*,pcm*,rawvideo,spdif,*pipe,webm,wav,dash,nu*'
15:36:45 [DEBUG] (10) mdk: [MDK] avdevice_register_all00007FFCC15B1000
15:36:45 [DEBUG] (10) mdk: [MDK] try to load av module: D:\Downloads\Gyroflow-windows64\ffmpeg-5.dll
15:36:45 [DEBUG] (10) mdk: [MDK] try to load av module: D:\Downloads\Gyroflow-windows64\avformat-59.dll
15:36:45 [DEBUG] (10) mdk: [MDK] Selected avformat runtime version: 59.16.100 (build: 59.17.102), license: GPL version 2 or later
15:36:45 [DEBUG] (10) mdk: [MDK] Selected avformat runtime configuration: --extra-version=QtAV --disable-doc --disable-debug --enable-shared --enable-runtime-cpudetect --enable-libmfx --disable-ptx-compression --enable-mediafoundation --disable-postproc --ranlib=llvm-ranlib-13 --nm=llvm-nm-13 --ar=llvm-ar-13 --cc=clang-13 --ld='/usr/lib/llvm-13/bin/lld-link ' --nvcc=clang-13 --enable-cross-compile --arch=x86_64 --target-os=win32 --disable-stripping --extra-cflags='--target=x8
15:36:45 [DEBUG] (10) mdk: [MDK] 6_64-pc-windows-msvc -DWIN32 -D_WIN32 -D_WIN32_WINNT=0x0600 -Wno-nonportable-include-path -Wno-deprecated-declarations -I/tmp/dep/include -Xclang -ivfsoverlay -Xclang "/tmp/winsdk/vfs.yaml" -g -gcodeview -Xclang -cfguard' --extra-ldflags='-DEBUG -OPT:REF -SUBSYSTEM:CONSOLE -NODEFAULTLIB:libcmt -DEFAULTLIB:msvcrt -guard:cf' --extra-libs=oldnames.lib --enable-small --disable-filters --enable-filter='*null*,*fade,*fifo,*format,*resample,aeval,all
15:36:45 [DEBUG] (10) mdk: [MDK] *,atempo,pan,color*,convolution,crop,draw*,eq*,framerate,*_cuda,*_qsv,*_vaapi,*v4l2*,hw*,scale,volume' --disable-muxers --disable-encoders --disable-decoders --enable-decoder='*sub*,movtext,*web*,aac*,*ac3*,alac*,ape,ass,av1*,cc_dec,cook,dca,dnxhd,eac3*,exr,truehd,ff*,*yuv*,flv,flac,gif,h26[3-4]*,hevc*,hap,mp[1-3]*,prores,*peg*,mlp,mpl2,nellymoser,opus,pcm*,*png*,rawvideo,rv*,sami,srt,ssa,v210*,vc1*,vorbis,vp[6-9]*,wm*,wrapped_avframe' --disab
15:36:45 [DEBUG] (10) mdk: [MDK] le-demuxers --enable-demuxer='*sub*,*ac3,*ac,*peg*,*web*,ape,ass,avi,concat,dnxhd,dts*,*dash*,*flv,gif,hls,h264,hevc,kux,xv,matroska,mlv,mov,mp3,mxf,nsv,nut,ogg,pcm*,rawvideo,rt*p,spdif,srt,vc1,v210*,wav,*pipe,image2' --disable-parsers --enable-parser='*sub*,aac*,ac3,cook,dnxhd,flac,h26[3-4],hevc,m*,opus,rv*,vc1,vorbis,vp[8-9]' --pkg-config=pkg-config --pkg-config-flags=--static --enable-gpl --enable-libx265 --enable-libx264 --enable-encoder='
[... 214 lines cut ...]
15:36:46 [DEBUG] (14) mdk: [MDK] crop: 0.9881 x 0.9727
15:36:46 [DEBUG] (14) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
15:36:46 [INFO] surface size: 1920x1080
15:36:46 [INFO] [MDK] seek flag: 2, from start: 1, to: 0, 0: 0, now: 0, key: 0, back: 0
15:36:46 [DEBUG] (14) mdk: [MDK] seeking media's position is 0=====
15:36:46 [INFO] [MDK] __int64 __cdecl mdk::abi::PacketIOWrapper::seek(__int64,enum mdk::abi::SeekFlag)@501 seek target(from 0): 0ms, flags: +FromStart
15:36:46 [DEBUG] (10) mdk: [MDK] Seek start. target time: 0 avseek flags: 1
15:36:46 [DEBUG] (10) mdk: [MDK] Seek end
15:36:46 [DEBUG] (10) mdk: [MDK] 1 packets is read after seek. read more to get target pts
15:36:46 [DEBUG] (10) mdk: [MDK] 1 packets is read after seek. seek result pts: 0s, requested: 0
15:36:46 [DEBUG] (10) mdk: [MDK] seek_drop_non_video_: 0 ms, seek_wait_video_frame_: 1...
[... cut ...]
15:36:46 [DEBUG] (13) mdk: [MDK] subtitle stream#2 AOT frame is sent
15:36:46 [DEBUG] (12) mdk: [MDK] >>>>>>>>1st audio frame (after seek) rendered: 1, ao: 0, a: 0, delta: 0 +0.021333
15:36:46 [DEBUG] (12) mdk: [MDK] audio stream#1 AOT frame is sent
15:36:46 [DEBUG] (11) mdk: [MDK] video stream#0 sending 1 invalid AOT frame @0.000000s. seeking: 0
15:36:46 [DEBUG] (11) mdk: [MDK] VideoRenderer clear buffered frames
15:36:46 [DEBUG] (11) mdk: [MDK] 0-track seek end video frame @0 seek_pos_: -1
15:36:46 [DEBUG] (11) mdk: [MDK] video stream#0 AOT frame is sent
```

AdrianEddy, 2022-03-11:
Please also paste full log from console, when you open Gyroflow_with_console.bat

rodrigosclosa, 2022-03-11:
> Please also paste full log from console, when you open Gyroflow_with_console.bat

17:37:27 [INFO] Latest version: v1.0.0-rc5, current version: 1.0.0-rc5
17:37:27 [INFO] Lens profiles directory: "\\\\?\\D:\\Downloads\\Gyroflow-windows64\\camera_presets"
17:37:27 [INFO] Loaded 1293 lens profiles in 88.289ms
17:37:28 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("NVIDIA CUDA") ["NVIDIA GeForce GTX 1060"]
17:37:28 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Intel(R) OpenCL HD Graphics") ["Intel(R) HD Graphics 630"]
17:37:28 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 1060
17:37:28 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
17:37:28 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src\rendering\mod.rs:47] gpu_type = NVIDIA
17:37:37 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
17:37:37 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.video" ""
17:37:37 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
17:37:37 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "thread.subtitle" ""
17:37:37 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
17:37:37 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "decoder.subtitle" "FFmpeg"
17:37:37 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
17:37:38 [DEBUG] (14) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
17:37:38 [DEBUG] (14) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x2950e193880) QSize(1124, 633)
17:37:38 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.D3D11"
17:37:38 [DEBUG] (14) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
17:37:38 [INFO] surface size: 1920x1080
Pressione qualquer tecla para continuar. . .

mycosd, 2022-03-23:
@AdrianEddy I can reproduce it, RC5 and current builds just close after dragging in the video.

mycosd, 2022-03-23:
We are missing `ImageFovType` 7 here: https://github.com/AdrianEddy/telemetry-parser/blob/141e94393bbd3f25e3eafc6b6f8d846ecd5708f1/src/insta360/extra_info.rs#L236 which is causing the crash

AdrianEddy, 2022-03-24:
@mycosd thanks for the investigation, I added it now, should be good in latest dev build: https://gyroflow.xyz/devbuild/?autodownload

closed 2022-03-24 by AdrianEddy