gyroflow/gyroflow #269: Option to Crop the output to a desired size

rodrigosclosa, 2022-03-11:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I've been using gyroflow to generate videos for a real time cover of events... It's a fantastic and practical tool for real time!

It would be very useful if there was a way to cut the video in a custom format, for example: I have a video recorded on the GoPro in 4K horizontally, but I could stabilize it in gyroflow and cut only a central track to export in the standard from Reels for instagram.

My suggestion would be to add those resize controls on top of the video player layer with some predefined presets, like for Reels.

AdrianEddy, 2022-03-11:
You can already export any resolution including vertical, even if input is horizontal, is this way you want?

rodrigosclosa, 2022-03-11:
> You can already export any resolution including vertical, even if input is horizontal, is this what you want?

Yes, I see that I can export in other resolcutions... But it's the entire video frame, not a cropped one... 
On a 4K 4:3 horizontal video, we can export 3 lanes in vertical on 1080p for example, to centralize what portion of video we want on final export... For vertical videos, it's a very usefull feature...

AdrianEddy, 2022-03-11:
You can use FOV slider to zoom in or out to get the desired frame.

rodrigosclosa, 2022-03-11:
> You can use FOV slider to zoom in or out to get the desired frame.

But we cannot change the video portion that we want, to adjust the desired information we want on final export...
Adding a crop option, adds the ability to select any portion of the video...

Entropy512, 2022-03-11:
Are you meaning an offset crop (not centered)?

rodrigosclosa, 2022-03-11:
> Are you meaning an offset crop (not centered)?

Yeah! It'll be amazing if we can scrool horizontally and choose the portion of video we want on output...

closed 2022-04-19 by AdrianEddy