gyroflow/gyroflow #270: Runcam Thumb gyro issue figured out

CREASE-gum-EAR, 2022-03-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

r5

### What operating system are you using?

Windows

### What GPU are you using?

RTX 2060

### What happened?

I figured out why the gyro data for the Runcam Thumb doesn't work. All they did was accidentally swapped the gyro and accelerometer data in the gcsv. So gyroflow was reading accell data and not gyro data. I opened the gcsv in excel, swapped the rx,ry,rz columns with the ax,ay,az columns, saved it, and then opened the file in notepad and fixed the top lines to get rid of the extra commas that excel added, and it worked. Perfectly stable footage. I know runcam should fix it, but could we temporarily fix it in gyroflow? I also updated the Thumb to the current firmware version 2.0 and it didn't fix it.

### Relevant log output

```shell
Original bad data:
GYROFLOW IMU LOG
version,1.1
id,runcam_thumb
orientation,XYZ
note,development_tanky
fwversion,Thumb V2.0.0
vendor,runcam
videofilename,Thumb0003.MP4
lensprofile,Runcam_Thumb_1080p_16by9
tscale,0.001
gscale,0.00054541539
ascale,0.00024414062
t,rx,ry,rz,ax,ay,az
1,-4239,-14,840,-405,81,22
2,-4239,-14,840,-405,81,22
3,-4238,-13,836,-409,70,17
4,-4242,-12,826,-412,70,12
5,-4242,-12,826,-412,70,12

Fixed Data
GYROFLOW IMU LOG
version,1.1
id,runcam_thumb
orientation,XYZ
note,development_tanky
fwversion,Thumb V2.0.0
vendor,runcam
videofilename,Thumb0003.MP4
lensprofile,Runcam_Thumb_1080p_16by9
tscale,0.001
gscale,0.00054541539
ascale,0.00024414062
t,rx,ry,rz,ax,ay,az
1,-405,81,22,-4239,-14,840
2,-405,81,22,-4239,-14,840
3,-409,70,17,-4238,-13,836
4,-412,70,12,-4242,-12,826
5,-412,70,12,-4242,-12,826
```

ElvinC, 2022-03-12:
Closing this as I already discussed this with the Runcam team the other day, expect a new firmware in the coming days

closed 2022-03-12 by ElvinC

CREASE-gum-EAR, 2022-03-12:
Sorry to bother you all then. This software is amazing!!!!!! Thank you so much.