gyroflow/gyroflow #271: FFmpeg error 12912

Starlights74, 2022-03-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

from 1.0.0rc3 to 1.0.0rc5

### What operating system are you using?

Mac OS Monterey 12.2.1

### What GPU are you using?

Radeon Pro 580 8Gbyte

### What happened?

it happens immediately when I press export
[image]

dustin, 2022-03-12:
I had a similar, but not identical error.  Disabling GPU encoding made it work for me.

[image]

The AAC error goes away if I disable audio, of course, but the render only proceeds with GPU disabled.

Starlights74, 2022-03-12:
thanks, without "GPU encoding" it works, but I lose all benefit of my iMac Radeon 580 Pro card.
on a i7 4,2Ghz it encoding only a 4fps a GoPro 4k 50fps file. :-(

AdrianEddy, 2022-04-13:
Please try latest dev build https://gyroflow.xyz/devbuild/?autodownload and upload the log file after you get the error and close the app

Open package contents -> Contents -> MacOS -> gyroflow.log file

AdrianEddy, 2022-04-30:
It should be fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-04-30 by AdrianEddy