gyroflow/gyroflow #272: FFmpeg error22

rolandau2013, 2022-03-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0-rc5 (gh378)

### What operating system are you using?

OS 10.14.6

### What GPU are you using?

AMD FirePro D500 3 GB

### What happened?

[image]
export error

AdrianEddy, 2022-03-15:
Open package contents -> Contents -> macOS -> drag gyroflow file into terminal and press enter. Then after you close the app, upload gyroflow.log file

rolandau2013, 2022-03-15:
is it ok??

rolandau2013, 2022-03-15:
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8251044/gyroflow.log)

AdrianEddy, 2022-04-30:
Fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-04-30 by AdrianEddy