gyroflow/gyroflow #274: ProRes HQ exported files, from a Sony RX0ii camera cant be read in premiere

dronotron, 2022-03-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0 rc5 latest build march 15

### What operating system are you using?

mac os Big Sur

### What GPU are you using?

Mac M1 

### What happened?

Exporting in ProRes HQ, LT, STD, 444 (havent tried in proxy) using a sony rx0ii file. it cant be opened  in premiere, sometimes it shows something with yellow and magenta colors, sometimes it gives a frame error.
If the file is exported on h.264 and h.265 it can be seen fine.

This only happes with the sony camera (any file recorded with the camera), if I export a file from the gopro, premiere opens it just fine.

sample file: https://drive.google.com/file/d/1yChIQBVCRifXMJS0gZ-5TXCcAmmUzCuW/view?usp=sharing
[gyroflow.log](https://github.com/gyroflow/gyroflow/files/8259273/gyroflow.log)
[image]

AdrianEddy, 2022-03-16:
Please also upload original video file

AdrianEddy, 2022-03-16:
Did you try unchecking GPU encoding? or GPU decoding?

dronotron, 2022-03-16:
> 

yes both gpu and software decoding. With gpu decoding sometimes it gives a ffmpeg error , this happens ramdomly

dronotron, 2022-03-16:
> Please also upload original video file

here is an old file from the same camera, uploaded previously. I tested with diffrent files from this same camera, all gives the same problem https://drive.google.com/file/d/1mk6BK5HeE2-OmHuShmnsgC365mhe-j5q/view?usp=sharing

dronotron, 2022-03-16:
The exported files play normally in quicktime , but they are unusable in premiere

AdrianEddy, 2022-11-18:
Does this still happen in latest dev build? https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-11-21:
I believe this is no longer an issue in latest dev build and updated Premiere, please reopen if it still happens https://gyroflow.xyz/devbuild/?autodownload

closed 2022-11-21 by AdrianEddy