gyroflow/gyroflow #275: Gyroflow 1.0.0 RC5 not responding or unresponsive during Rendering/export, once Crashes or Freeze. Win 10 Pro.

gonebdg, 2022-03-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0.0-rc5 (gh379)

### What operating system are you using?

Edition: Windows 10 Pro, Version: 21H2, OS build: 19044.1586, Experience: Windows Feature Experience Pack 120.2212.4170.0

### What GPU are you using?

Intel(R) HD Graphics (on board)

### What happened?

**Computer**
Processor : Intel(R) Core(TM) i5-3330 CPU @ 3.00GHz   3.00 GHz
Installed RAM : 8,00 GB (7,46 GB usable)
System type : 64-bit operating system, x64-based processor
.
**Action Camera** 
Brand/Type : Runcam Thumb with FW v2.1.0
Video : 1080p50fps (for detail media info, please check the video file directly)
.
**Descriptions**
No problem during insert video, settings, or play video/preview in Gyroflow UI.
No problem export to .gyroflow file
But then Gyroflow not responding or unresponsive during Rendering/export stabilize video, once Crashes or Freeze, once I have to powering off my computer and restart because my computer is not responding at all.
Please watch this screen recording :  [Gyroflow_gulNg7hoEM.mp4](https://drive.google.com/file/d/1EPWQo1AK2cMdMgleUfCHnPmGOydrG6-b/view?usp=sharing)
.
**Files to reproduce the issue (google drive):**

1. [Thumb0004.MP4](https://drive.google.com/file/d/1V1aj95YG8RAjmtm84LN8JypVUyjvEtkb/view?usp=sharing)
2. [Thumb0004.gcsv](https://drive.google.com/file/d/1A_b2a8-KUIbUibHYPq2R_bk-Y8FNTvQq/view?usp=sharing)
3. [gyroflow.log](https://drive.google.com/file/d/1SytO-Y4uEN3o2Kmzy-k58iGUkT-pvx52/view?usp=sharing) (last try)

### Relevant log output

```shell
n/a
please see the gyroflow.log attached
```

AdrianEddy, 2022-03-19:
Does it work when you disable GPU encoding?

gonebdg, 2022-03-19:
> Does it work when you disable GPU encoding?

I've tried but it doesn't work with the same issue.

gonebdg, 2022-03-24:
> I've tested it on Win 10 Pro Version 21H1 with the exact same settings you showed in the screen recording. Works fine here. As you are not using GPU for decoding and encoding and the preview also works fine and it also happens so quick after hitting export, my guess would be that there is something on your system blocking Gyroflow from writing to disk. Maybe virus scanner waiting for confirmation (I know of AVG doing this)? Try a different export directory (eg. your user/video and user/documents directory should be always unblocked). Try temporary disabling virus scanner. Try running Gyroflow as admin.

Hi @mycosd 
Thank you for collaborating to this issue. But my Win 10 Pro Version is **21H2** instead 21H1 and the OS build is: **19044.1586** and I think this is the source of the problem.
Since I use Win 10, I didn't use 3rd party Anti Virus software anymore and only rely on Window Security/Defender

Hudiny27, 2022-03-24:
[image]
Same problem with RC5 on my computer. After I hit Render->The Program freezes, not responding. Tried to wait but after an hour the same window and progress are at 0%. RC4 was running well.
Tried to feed the program with GoPro Session 5, and Hero 9 footage, but the result is the same.
OS: Win 10 Pro, 19044.1586

AdrianEddy, 2022-03-24:
@Hudiny27 Please also attach gyroflow.log

AdrianEddy, 2022-03-24:
Please try latest dev build: https://gyroflow.xyz/devbuild/?autodownload and let me know

Hudiny27, 2022-03-25:
This latest DEV build works great! Thanks!

gonebdg, 2022-03-25:
@Hudiny27 could you please share your settings and your windows environtment ?
@AdrianEddy the latest build (gh391) still not work for me :(
try with different export path location, try execute gyroflow with "Run as Administrator", also try to put the gyroflow in my windows 10 system drive C:\Program Files\Gyroflow but no luck, I still have the unresponsive issue.
check following video and the log (last try exec. >  C:\Program Files\Gyroflow > run as administrator) 
in my google drive : [https://drive.google.com/drive/folders/19Gw-taBKcvYYjFXviKM81VmsCb-Jw531?usp=sharing](https://drive.google.com/drive/folders/19Gw-taBKcvYYjFXviKM81VmsCb-Jw531?usp=sharing)

gonebdg, 2022-03-25:
@Hudiny27 please reopen this issue, the latest build still not work for me.

gonebdg, 2022-03-25:
@mycosd I think you close this issue thread not @Hudiny27 (sorry)
Please reopen this iisue.

mycosd, 2022-03-25:
If Gyroflow using any CPU while it is just doing nothing?

AdrianEddy, 2022-03-25:
It appears the issue is in OpenCL when using Intel HD Graphics on some devices (I have one device with that integrated graphics and it works in my case).
@Hudiny27 also had NVIDIA graphics and rc5 wrongly chosen Intel HD Graphics, which caused the problem. In dev build it correctly chosen NVIDIA one and that solved it for him

@gonebdg however has Intel HD Graphics only 

@gonebdg Can try try this: Edit `Gyroflow_with_console.bat` and add this as a second line:
`set NO_OPENCL=1`
then run Gyroflow_with_console.bat

if that doesn't help, also add `set NO_WGPU=1` as a third line

gonebdg, 2022-03-25:
@AdrianEddy Yes now it works!!
By adding the `set NO_OPENCL=1` only is works but I saw errors:
`thread '<unnamed>' panicked at 'wgpu error: Validation Error` ... etc.
When I add the `set NO_WGPU=1` those errors gone.
However for both conditions I saw another errors in console when I start "Auto sync"
`[ERROR] OpenCV error: Error { code: 0, message: "Model not found" }`

May I know what is the implication of parameter `set NO_OPENCL=1` and the `set NO_WGPU=1` to the rendering process and the result ?

AdrianEddy, 2022-03-25:
It just disables GPU processing and does everything on the CPU.

You may also try to update your graphics drivers and install all windows updates, it's hard to tell what's the exact issue there.
You can also send me the gyroflow.log with that `thread '<unnamed>' panicked at 'wgpu error: Validation Error` error, maybe it'll have some useful information

gonebdg, 2022-03-25:
@AdrianEddy here is the full console log after I've try to render 3minutes video in 1hour
My computer spec. is too low for this process but it's ok ... the rendering time still acceptable for me :)
[gyrowflow.full-render.console.log.txt](https://drive.google.com/file/d/1tfqkz4jSkTx30KsKCqSVHei7hE2epAYL/view?usp=sharing).
Thank you for all of your hard work.

fremebondo, 2022-03-25:
@AdrianEddy , I am face the same issue on a dual-graphic-card laptop (Intel+Nvidia). Is there an option to force Gyroflow to use one GPU or the other? I suspect it uses Intel, even if I set up Gyroflow.exe to be served by NVIDIA card from NVIDIA control panel.

AdrianEddy, 2022-03-25:
@fremebondo use latest dev build, it should detect nvidia correctly. In any case you can upload gyroflow.log and I can check which one it used

https://gyroflow.xyz/devbuild/?autodownload

gonebdg, 2022-03-27:
Since running the gyroflow console bat with specific parameters work for me, I decide to close this thread.
Watch the stabilized video on my instagram
https://www.instagram.com/reel/Cbka1f2hVd5/?utm_medium=copy_link
Thanks!

closed 2022-03-27 by gonebdg