gyroflow/gyroflow #281: No_Image

EladZa, 2022-03-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0

### What operating system are you using?

win 10

### What happened?

After clicking on an EXE file, an empty window opens and there is no interface on it.
Just a blank screen.

AdrianEddy, 2022-03-24:
Update your NVIDIA drivers

AdrianEddy, 2022-03-24:
Can you try latest dev build?
https://gyroflow.xyz/devbuild/?autodownload

If that doesn't help, try different GUI backend by editing `Gyroflow_with_console.bat` and adding this as a second line:
`set QSG_RHI_BACKEND=opengl`
or
`set QSG_RHI_BACKEND=vulkan`

EladZa, 2022-03-24:
New problem...
gyroflow.exe - entry point not found

AdrianEddy, 2022-04-13:
Closing this as it's most likely something wrong with your system rather than Gyroflow. Make sure you have latest GPU drivers, maybe looks for DirectX updates or something like that. Other than that the `set QSG_RHI_BACKEND=opengl` and `vulkan` options should allow you to run it. There's nothing else we can do about it here

closed 2022-04-13 by AdrianEddy