gyroflow/gyroflow #284: Can t open gyroflow

teletof, 2022-03-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

The last one

### What operating system are you using?

Windows 10

### What happened?

When i try to start gyroflow i have an error 0xc000007b.
The only version that works is the bêta version.
Thank you for your great work.

AdrianEddy, 2022-03-25:
did you install https://aka.ms/vs/17/release/vc_redist.x64.exe ?

teletof, 2022-03-25:
Yes or is done to.

teletof, 2022-03-25:
[image]

AdrianEddy, 2022-03-25:
Also install all windows updates and latest gpu drivers

teletof, 2022-03-25:
yes everything is ok

AdrianEddy, 2022-03-25:
so is gyroflow working now?

teletof, 2022-03-25:
no the same thing

AdrianEddy, 2022-03-25:
how about 
Settings -> Apps -> Optional features -> Add a feature -> enable Media Feature Pack

teletof, 2022-03-25:
it is ok and still not working

AdrianEddy, 2022-04-13:
Closing this as it's most likely something wrong with your system rather than Gyroflow. You can google for the errors you're getting, they are system errors.

closed 2022-04-13 by AdrianEddy