gyroflow/gyroflow #286: Gyroflow stuck on rendering

matx93, 2022-03-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0-rc5

### What operating system are you using?

Windows 10 19042

### What GPU are you using?

GT740M (but tried also without)

### What happened?

Tried to stabilize a drone shooting from a Caddx Loris 4k through betaflight v4.3.0 blackbox log.
Sync and stabilization look ok in Gyroflow v1.0.0-rc5 but the export rendering is stuck at 0% even after hours.
Tried with/without GPU rendering. 
Latest redist installed

Uploading video and bbl files here: 
https://1drv.ms/u/s!An6FA_wrlvnlsQ1DW0LH_g47BbaK?e=XlXjKj

mycosd, 2022-03-29:
This should be already fixed, see #275 
Please try the latest dev-build: https://gyroflow.xyz/devbuild/?autodownload

matx93, 2022-03-29:
Indeed, problem solved! Thank you

closed 2022-03-29 by matx93